ibaiGorordo/pyKinectAzure #1: Invalid argument to image_dec_ref()

lsding, 2020-06-29:
[2020-06-29 14:26:57.835] [error] [t=16976] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     16077 type:Depth
[2020-06-29 14:26:57.871] [error] [t=16976] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     49422 type:Depth
[2020-06-29 14:26:57.895] [error] [t=24284] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2020-06-29 14:26:57.895] [error] [t=24284] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (363): Invalid argument to image_dec_ref(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t

ibaiGorordo, 2020-06-29:
Hello,

I have never seen that error, so could you provide more information:

- SDK version

- What example program were you running.

- Did the error happened at the beginning? Or were you able to receive some data first and then the error occurred?

- Is that all the error that is displayed in the console?

- Are you using more than one Kinect camera?

Also, a quick search into the error led me to t[his issue in the official SDK repository](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1112). I am not sure if it is the same problem, does the official Azure Kinect Viewer program work well in your computer?

Best regards,
Ibai

lsding, 2020-06-30:
Hi，
1、SDK version 1.4.0，only one Kinect
2、It happened all the example program，but I can get Depth image and IR image，can‘t get color image
3、Error happened at the beginning，and k4aviewer work well in my computer

Thanks 

 python exampleDepthImageOpenCV.py
[2020-06-30 09:58:00.057] [error] [t=24216] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    182755 type:Depth
[2020-06-30 09:58:00.257] [error] [t=24216] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    382755 type:Depth
[2020-06-30 09:58:00.458] [error] [t=24216] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    582755 type:Depth
[2020-06-30 09:58:00.658] [error] [t=24216] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    782755 type:Depth
[2020-06-30 09:58:00.857] [error] [t=24216] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    982755 type:Depth
[2020-06-30 09:58:01.058] [error] [t=24216] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1182755 type:Depth
[2020-06-30 09:58:01.258] [error] [t=24216] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1382755 type:Depth
[2020-06-30 09:58:01.459] [error] [t=24216] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1582755 type:Depth
[2020-06-30 09:58:01.659] [error] [t=24216] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1782755 type:Depth
[2020-06-30 09:58:01.860] [error] [t=24216] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1982755 type:Depth
Device configuration:
        color_format: 0
        (0:JPG, 1:NV12, 2:YUY2, 3:BGRA32)

        color_resolution: 2
        (0:OFF, 1:720p, 2:1080p, 3:1440p, 4:1536p, 5:2160p, 6:3072p)

        depth_mode: 2
        (0:OFF, 1:NFOV_2X2BINNED, 2:NFOV_UNBINNED,3:WFOV_2X2BINNED, 4:WFOV_UNBINNED, 5:Passive IR)

        camera_fps: 0
        (0:5 FPS, 1:15 FPS, 2:30 FPS)

        synchronized_images_only: False
        (True of False). Drop images if the color and depth are not synchronized

        depth_delay_off_color_usec: 0 ms.
        Delay between the color image and the depth image

        wired_sync_mode: 0
        (0:Standalone mode, 1:Master mode, 2:Subordinate mode)

        subordinate_delay_off_master_usec: 0 ms.
        The external synchronization timing.

        disable_streaming_indicator: False
        (True or False). Streaming indicator automatically turns on when the color or depth camera's are in use.

lsding, 2020-06-30:
It's not bug of the API,
It works well when I debug it in Pycharm, but can't get color_image_handle when run it in Pycharm or in console.
I

ibaiGorordo, 2020-06-30:
Thank you for providing the information.

I have no idea why that could happen. In my case I use the Anaconda command prompt and it works fine, but it it really weird that it works in Pycharm only in the debug mode. Sorry, I cannot help you.

Let me know if you have any other problem/comment/questions.

Best regards,
Ibai

closed 2021-03-03 by ibaiGorordo