ibaiGorordo/pyKinectAzure #2: transformation_depth_image_to_color_camera

lsding, 2020-07-01:
Hi:
    I want to transform depth_image to color camera, Can you give me example?
Thanks

ibaiGorordo, 2020-07-01:
Hi,

I have just added an example that uses the transformation of the depth to the color format. The example overlays the transformed depth color map (**transformed_depth_color_image**) over the color image (**color_image**). 

If you just want to get the transformed depth image you can use the **transformed_depth_image** variable.

I also found that it took more time to process each sample, I am not sure if the performance can be improved.

Let me know how it goes.

Ibai

lsding, 2020-07-02:
Hi，
It works well.
Thank you very much

closed 2020-07-02 by lsding