ibaiGorordo/pyKinectAzure #3: How to use device_get_serialnum

lsding, 2020-07-02:
Hi,
I want to get device serialnum,but got this error.
What's wrong about it?   I have no idea about ctypes.

 def device_get_serialnum(self):
        serial_number = 0
        serial_number_size = 0
        self.k4a.k4a_device_get_serialnum(self.device_handle,serial_number,serial_number_size)

[2020-07-02 15:43:50.392] [error] [t=23504] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (292): Invalid argument to depth_get_device_serialnum(). serial_number_size == NULL
[2020-07-02 15:43:50.392] [error] [t=23504] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (292): depth_get_device_serialnum() returned failure.
[2020-07-02 15:43:50.392] [error] [t=23504] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (962): depth_get_device_serialnum(device->depth, serial_number, serial_number_size) returned failure in k4a_device_get_serialnum()

lsding, 2020-07-03:
def device_get_serialnum(self):
        serial_number = None
        serial_number_size = ctypes.c_size_t()
        result = self.k4a.k4a_device_get_serialnum(self.device_handle, serial_number, serial_number_size)
        if result == _k4a.K4A_BUFFER_RESULT_TOO_SMALL:
            serial_number = ctypes.create_string_buffer(serial_number_size.value)
            result = self.k4a.k4a_device_get_serialnum(self.device_handle, serial_number, serial_number_size)

closed 2020-07-03 by lsding

ibaiGorordo, 2020-07-03:
Hi,

Sorry for the late response. 

Thank you for working out the solution. I have modified your code a bit and added to the repository.

Let me know if you have any other suggestions, I will look at it once I have time.

Ibai