ibaiGorordo/pyKinectAzure #4: would you add record and playback function？

lsding, 2020-07-07:
Hi：
I added record_create function in _k4a.py, but got error，function not found
would you add record and playback function please？Thank You

```
        self.k4a_record_create = dll.k4a_record_create
        self.k4a_record_create.restype = k4a_result_t
        self.k4a_record_create.argtypes = (ctypes.POINTER(ctypes.c_char), \
                                           k4a_device_t, \
                                           k4a_device_configuration_t, \
                                           k4a_record_t, \
                                           )
```

```
  File "../pyKinectAzure\pyKinectAzure.py", line 14, in __init__
    self.k4a = _k4a.k4a(modulePath)
  File "../pyKinectAzure\_k4a.py", line 560, in __init__
    self.k4a_record_create = dll.k4a_record_create
  File "D:\soft\Anaconda3\lib\ctypes\__init__.py", line 369, in __getattr__
    func = self.__getitem__(name)
  File "D:\soft\Anaconda3\lib\ctypes\__init__.py", line 374, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'k4a_record_create' not found
```

ibaiGorordo, 2020-07-08:
Hi,

I had not considered to add playback. Is there any benefit on using the built in record function instead of saving the images using OpenCV?

However, I think adding playback to visualize already recorded data could be interesting. For example, record the data using the visualizer in the Kinect Azure SDK and then read that playback in python to perform some processing. I will look into that this weekend if I have time.

Ibai

lsding, 2020-07-10:
Hi
I have already resolved it,
record and playback functions are in  k4arecord.dll.
Thank you for your reply.

ibaiGorordo, 2020-07-10:
I am glad to hear that. Could you add a link to your code?

It might help other people that want that also.

Ibai

lsding, 2020-08-18:
class k4a:

    def __init__(self, modulePath):
        try:
            dll = ctypes.CDLL(modulePath)
            recorddll = ctypes.CDLL(modulePath.replace('k4a.dll', 'k4arecord.dll'))
        except Exception as e:
            try:
                dll = ctypes.CDLL('k4a.so')
                recorddll = ctypes.CDLL('k4arecord.so')
            except Exception as ee:
                print("Failed to load library", e, ee)
                sys.exit(1)

        self.k4a_record_create = recorddll.k4a_record_create
        self.k4a_record_create.restype = k4a_result_t
        self.k4a_record_create.argtypes = (ctypes.POINTER(ctypes.c_char), \
                                           k4a_device_t, \
                                           k4a_device_configuration_t, \
                                           ctypes.POINTER(k4a_record_t),\
                                           )

        self.k4a_record_write_header = recorddll.k4a_record_write_header
        self.k4a_record_write_header.restype = k4a_result_t
        self.k4a_record_write_header.argtypes = (k4a_record_t,)

        self.k4a_record_write_capture = recorddll.k4a_record_write_capture
        self.k4a_record_write_capture.restype = k4a_result_t
        self.k4a_record_write_capture.argtypes = (k4a_record_t, \
                                                  k4a_capture_t)

        self.k4a_record_flush = recorddll.k4a_record_flush
        self.k4a_record_flush.restype = k4a_result_t
        self.k4a_record_flush.argtypes = (k4a_record_t,)

        self.k4a_record_close = recorddll.k4a_record_close
        self.k4a_record_close.restype = None
        self.k4a_record_close.argtypes = (k4a_record_t,)

        self.k4a_playback_open = recorddll.k4a_playback_open
        self.k4a_playback_open.restype = k4a_result_t
        self.k4a_playback_open.argtypes = (ctypes.POINTER(ctypes.c_char), \
                                           ctypes.POINTER(k4a_playback_t),)

        self.k4a_playback_close = recorddll.k4a_playback_close
        self.k4a_playback_close.restype = None
        self.k4a_playback_close.argtypes = (k4a_playback_t,)

        self.k4a_playback_get_next_capture = recorddll.k4a_playback_get_next_capture
        self.k4a_playback_get_next_capture.restype = k4a_stream_result_t
        self.k4a_playback_get_next_capture.argtypes = (k4a_playback_t, \
                                                       ctypes.POINTER(k4a_capture_t),)

        self.k4a_playback_get_calibration = recorddll.k4a_playback_get_calibration
        self.k4a_playback_get_calibration.restype = k4a_result_t
        self.k4a_playback_get_calibration.argtypes = (k4a_playback_t, \
                                                    ctypes.POINTER(k4a_calibration_t), \
                                                    )

ibaiGorordo, 2020-08-18:
Thank you @lsding, I will try to test it this weekend and added it to the repository.

donghaoye, 2021-05-14:
> Thank you @lsding, I will try to test it this weekend and added it to the repository.

Hi, Did you add it to the repos?

ibaiGorordo, 2021-05-14:
Actualy

donghaoye, 2021-05-14:
> Actually, I did not. In order to test it I had to also modify other parts of the code.
> 
> I might give it a try now.

Great! Could you give some examples of playback and record?

ibaiGorordo, 2021-05-14:
Working on that now. I will try to upload them in a couple of hours.

donghaoye, 2021-05-14:
> Working on that now. I will try to upload them in a couple of hours.

Thank you so much. Looking forward to your good news.

donghaoye, 2021-05-14:
> Working on that now. I will try to upload them in a couple of hours.

Nice job! Very glad to see the recording example.
Could you add a playback example in the following hours? Thank you.

ibaiGorordo, 2021-05-14:
Working on that also, but it is a bit more nested with the k4a library. It might take a while.

donghaoye, 2021-05-17:
> Working on that also, but it is a bit more nested with the k4a library. It might take a while.

Thanks for your efforts.  Have you made any progress on it?

ibaiGorordo, 2021-05-17:
I have ended up doing a major refactor to the repository, I almost have finished but there are some parts that are not working correctly.

I will probably finish it this weekend (no time during the week).

egeozsoy, 2021-09-24:
Any updates here?

ibaiGorordo, 2021-10-04:
I am still busy with other projects, my recommendation for now is to use this other repository:
https://github.com/etiennedub/pyk4a/tree/master/pyk4a