ibaiGorordo/pyKinectAzure #5: This is SO GREAT

murez, 2020-07-30:
What an amazing python project and this can be used on ***linux*** too!
I have tested on **Ubuntu 20.04**, just modify the `modulePath` into `"/usr/lib/x86_64-linux-gnu/libk4a.so"` and it works smoothly just like on windows.

ibaiGorordo, 2020-07-30:
Hi,

Thank you for kind words. I am glad to hear it works well also in linux. I will update the Readme information about it.

Lately, I have been busy with other projects so I have not updated much, but if you have any desired feature or problem, please let me know.

Ibai

closed 2020-08-01 by murez

murez, 2020-08-01:
> Hi,
> 
> Thank you for kind words. I am glad to hear it works well also in linux. I will update the Readme information about it.
> 
> Lately, I have been busy with other projects so I have not updated much, but if you have any desired feature or problem, please let me know.
> 
> Ibai

BTW, under `Ubuntu 20.04` this only works with the root permission which means that when executing the `py` file, you must add `sudo` or make sure that the user has the right to the Azure Kinect DK, and you may talk about it in the `README` files.