ibaiGorordo/pyKinectAzure #9: help!AttributeError: undefined symbol: k4a_calibration_color_2d_to_depth_2d

Double-zh, 2020-08-07:
runfile('/home/zhengzh/DK/DK_C++/pyKinectAzure-master/examples/exampleDepthImageOpenCV.py', wdir='/home/zhengzh/DK/DK_C++/pyKinectAzure-master/examples')
Traceback (most recent call last):

  File "<ipython-input-1-e092061321bf>", line 1, in <module>
    runfile('/home/zhengzh/DK/DK_C++/pyKinectAzure-master/examples/exampleDepthImageOpenCV.py', wdir='/home/zhengzh/DK/DK_C++/pyKinectAzure-master/examples')

  File "/home/zhengzh/anaconda3/lib/python3.6/site-packages/spyder/utils/site/sitecustomize.py", line 705, in runfile
    execfile(filename, namespace)

  File "/home/zhengzh/anaconda3/lib/python3.6/site-packages/spyder/utils/site/sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/home/zhengzh/DK/DK_C++/pyKinectAzure-master/examples/exampleDepthImageOpenCV.py", line 16, in <module>
    pyK4A = pyKinectAzure(modulePath)

  File "../pyKinectAzure/pyKinectAzure.py", line 18, in __init__
    self.k4a = _k4a.k4a(modulePath)

  File "../pyKinectAzure/_k4a.py", line 481, in __init__
    self.k4a_calibration_color_2d_to_depth_2d = dll.k4a_calibration_color_2d_to_depth_2d

  File "/home/zhengzh/anaconda3/lib/python3.6/ctypes/__init__.py", line 361, in __getattr__
    func = self.__getitem__(name)

  File "/home/zhengzh/anaconda3/lib/python3.6/ctypes/__init__.py", line 366, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))

AttributeError: /home/zhengzh/Azure-Kinect-Sensor-SDK/build/bin/libk4a.so: undefined symbol: k4a_calibration_color_2d_to_depth_2d

ibaiGorordo, 2020-08-07:
Hi, 

Which Azure KInect SDK are you using? The problem might be caused because this function was not available in older SDK versions. If you have an older version (other than 1.4.X) try to update the SDK.

Let me know how it goes.

Ibai

Double-zh, 2020-08-08:
The Linux version I use is 16. 04, will it have an impact

> Hi,
> 
> Which Azure KInect SDK are you using? The problem might be caused because this function was not available in older SDK versions. If you have an older version (other than 1.4.X) try to update the SDK.
> 
> Let me know how it goes.
> 
> Ibai
HI，
The Linux version I use is 16. 04, will it have an impact？

ibaiGorordo, 2020-08-08:
Hi, 

The Linux version should not be a problem, what I mean is the version of the k4a-tools you have installed. My guess is that you have an older version of the k4a-tools that does not include the color_2d_to_depth_2d function. So if you can, try to update the k4a-tools (I am not sure how since I work with the Kinect on Windows).

Also, just to make sure, do the rest of the examples work?

Ibai

Double-zh, 2020-08-08:
well.
I also want to know how to install SDK under windows. Can you share some tutorials for me？

ibaiGorordo, 2020-08-08:
Check [this link](https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download). In windows it should be easy, you just need to download the Microsoft installer and run it to install the SDK.

Tell me how it goes.

Ibai

Double-zh, 2020-08-08:
OK, I'll tell you later！Thanks！

Double-zh, 2020-08-08:
Hi,the rest of the examples don't work!

zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ python exampleInfraredImageOpenCV.py
Traceback (most recent call last):
  File "exampleInfraredImageOpenCV.py", line 5, in <module>
    from pyKinectAzure import pyKinectAzure, _k4a
  File "../pyKinectAzure/pyKinectAzure.py", line 6, in <module>
    from config import config
  File "../pyKinectAzure/config.py", line 43
    f"\tcolor_format: {self.color_format} \n\t(0:JPG, 1:NV12, 2:YUY2, 3:BGRA32)\n\n"
    ^
SyntaxError: invalid syntax

ibaiGorordo, 2020-08-08:
I think that error is because you are using Python 2 which does not support f-Strings, try to run the demos using Python 3.

Ibai

Double-zh, 2020-08-08:
You're right, but the final output error is the same...

zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ python3 --version 
Python 3.5.2
zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ python2 --version   
Python 2.7.12
zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ python --version
Python 2.7.12
zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ echo alias python=python3 >> ~/.bashrc
zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ 
zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ source ~/.bashrc
zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ python --version
Python 3.5.2
zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ python exampleInfraredImageOpenCV.py
Traceback (most recent call last):
  File "exampleInfraredImageOpenCV.py", line 5, in <module>
    from pyKinectAzure import pyKinectAzure, _k4a
  File "../pyKinectAzure/pyKinectAzure.py", line 6, in <module>
    from config import config
  File "../pyKinectAzure/config.py", line 43
    f"\tcolor_format: {self.color_format} \n\t(0:JPG, 1:NV12, 2:YUY2, 3:BGRA32)\n\n"
    ^
SyntaxError: invalid syntax
zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ python exampleDepthImageOpenCV.py
Traceback (most recent call last):
  File "exampleDepthImageOpenCV.py", line 5, in <module>
    from pyKinectAzure import pyKinectAzure, _k4a
  File "../pyKinectAzure/pyKinectAzure.py", line 6, in <module>
    from config import config
  File "../pyKinectAzure/config.py", line 43
    f"\tcolor_format: {self.color_format} \n\t(0:JPG, 1:NV12, 2:YUY2, 3:BGRA32)\n\n"
    ^
SyntaxError: invalid syntax
zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$ python exampleColorImageOpenCV.py
Traceback (most recent call last):
  File "exampleColorImageOpenCV.py", line 5, in <module>
    from pyKinectAzure import pyKinectAzure, _k4a
  File "../pyKinectAzure/pyKinectAzure.py", line 6, in <module>
    from config import config
  File "../pyKinectAzure/config.py", line 43
    f"\tcolor_format: {self.color_format} \n\t(0:JPG, 1:NV12, 2:YUY2, 3:BGRA32)\n\n"
    ^
SyntaxError: invalid syntax
zhengzh@zhengzh:~/DK/DK_C++/pyKinectAzure-master/examples$

ibaiGorordo, 2020-08-08:
True, should be Python 3.6 or above, sorry.

Double-zh, 2020-08-08:
I deleted the 1.2 version of the SDK, and now I am installing the 1.4.0 version of the SDK, but I have encountered the following problems：
zhengzh@zhengzh:~/Azure-Kinect-Sensor-SDK/scripts/docker$ sudo ./setup-ubuntu.shsudo: ./setup-ubuntu.sh：找不到命令

So I would like to ask you how to install version 1.4, thank you！

ibaiGorordo, 2020-08-08:
Do you mean how to install the 1.4 SDK in Ubuntu? In the link I posted before there is information about it but I am not sure if it supports SDK 1.4. 

I have just seen that you also asked in Microsoft's Azure-Kinect-Sensor-SDK, hopefully, they can help you with it.

 Good luck!

Double-zh, 2020-08-08:
Yes, aren't you installing a 1.4.0 SDK?

ibaiGorordo, 2020-08-08:
Currently, I am using the 1.4.1 SDK but in Windows 10 (I also used the 1.4.0 in Windows). But I have not tried in Ubuntu. In [this issue](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1206), there is the link for the deb packages for the 1.4.1, however, I am not sure if that is useful for you.

Double-zh, 2020-08-08:
Thanks a lot！
Can you help me solve the following error？0.0..
zhengzh@zhengzh:~$ sudo apt install k4a-tools
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
有一些软件包无法被安装。如果您用的是 unstable 发行版，这也许是
因为系统无法达到您要求的状态造成的。该版本中可能会有一些您需要的软件
包尚未被创建或是它们已被从新到(Incoming)目录移出。
下列信息可能会对解决问题有所帮助：

下列软件包有未满足的依赖关系：
 k4a-tools : 依赖: libk4a1.4 (= 1.4.1) 但是它将不会被安装
             依赖: libc6 (>= 2.27) 但是 2.23-0ubuntu11.2 正要被安装
E: 无法修正错误，因为您要求某些软件包保持现状，就是它们破坏了软件包间的依赖关系。

If the following problem can not be solved, I can only install it under window

murez, 2020-08-09:
Well, I have met the same problem under ubuntu 20.04. It is because the dependent order of the two packages. So please install libk4a1.4 before k4a-tools.
我在ubuntu20.04下安装的时候也遇到这样的问题，是依赖顺序的问题，你应该先安装libk4a1.4，再安装 k4a-tools，所以现在你需要做的是把安装好的包卸载了。（有可能是我记反了，这个只有你自己去尝试或者到Microsoft那里去发issue

ibaiGorordo, 2020-08-09:
Thank you for the info!

Hopefully it helps with the problem.

closed 2020-10-28 by ibaiGorordo