ibaiGorordo/pyKinectAzure #13: Body tracking?

jts3k, 2021-01-07:
Does this library support body tracking?  The Readme seems to suggest so by suggesting that other similar solutions lack body tracking, and I see a branch to "add body tracking" but can't find any documentation for how to use body tracking, if indeed it is supported.

ibaiGorordo, 2021-01-08:
Hi,

Yes body tracking is definetely one of the things I want to implement. I have been busy with other projects so I have not done it yet. I will give it a try this weekend but I do not promise anything. 

Ibai

ibaiGorordo, 2021-01-08:
I have added the body tracking support. Still is in development but I have added an example to visualize the semantic segmentation body images over the depth image.

https://github.com/ibaiGorordo/pyKinectAzure/blob/master/examples/exampleSimpleBodyTracking.py

jts3k, 2021-01-08:
Thanks for the update!  I'll be eager to try this out.

ibaiGorordo, 2021-03-03:
Finally, I have just added the skeleton drawing in 2D.

I will close this issue.

Thank you for your patience and let me know if you have any other comments.

closed 2021-03-03 by ibaiGorordo

qbitronics, 2021-03-08:
Thats an awesome update. Thank you.
Would be intrested to know if bodyTracking supports aarch64 jetson nano? 
If yes, do point me the file. That would be Great.
Thanks in advance

qbitronics, 2021-03-08:
Hi,

Thats an awesome update. Thank you.

Would be intrested to know if bodyTracking supports aarch64 jetson nano?

If yes, do point me the file. That would be Great
.
Thanks in advance

ibaiGorordo, 2021-03-08:
Last time I checked I think there was no library for the body tracking in aarch64.

But I have not checked it recently, so it might have changed.

qbitronics, 2021-03-08:
Thank you for the early reply.

Yes I've checked on it, but couldn't find any library related to bodyTracking on aarch64.

Having a tough time as bodyTracking is a vital need for my project now.

Anyway do keep me posted if any updates on bodyTracking for aarch64. That would be a sigh of relief.

Thanks and regards.

ibaiGorordo, 2021-03-08:
I have heard of other users also with the same problem. If you do kot need the 3D pose to be very accurate, you could do pose estimation on the RGB image using some pose estimation model (Tflite posenet, openpose...) and then transform the 2D keypoints to 3D using the SDK function to convert from 2D to 3D.

Othewise, you might try to use the .onnx model directly without the library, my guess is that the library mainly converts the depth inage into some suitable format, performs the pose estimation, and parse the raw output to return the 3d pose.

Good luck with your project.

qbitronics, 2021-04-15:
Thank you so much for that. 
Proved useful for me.
And when can we expect the Python Point Cloud library  for Azure kinect? Any updates on that?

Thanks in Advance