ibaiGorordo/pyKinectAzure #15: Error while using exampleBodySegmentation.py

frankkim1108, 2021-01-25:
Hi I am currently working on a Azure Kinect project and found your library.
I tried using exampleBodySegmentation.py but I got a error saying 

```
[2021-01-25 13:51:02.859] [error] [t=23784] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (467): VerifyDependencies(). Cannot locate onnxruntime.dll
[2021-01-25 13:51:02.859] [error] [t=23784] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (129): Create(). At least one dependent library is missing!
[2021-01-25 13:51:02.859] [error] [t=23784] [K4ABT] D:\a\1\s\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
Body tracker initialization failed!

```
I have Azure Kinect Body Tracking SDK v1.0.0 and Azure Kinect SDK v1.4.1 installed at the correct path.
I added the files to the Path variable in Environment variable and it still didn't work.

It would be thankful if you could find the reasons to these problem.

Thank you for you time.

ibaiGorordo, 2021-02-01:
Hi,

Sorry for the late response, as you mentioned it seems that it is a path issue. Actually, I did not have to add the folder to the path to work in my computer. Does the "Azure Kinect Body Tracking Viewer"  program work?

frankkim1108, 2021-02-22:
Hi,
Sorry for the late response. I had to go to a 3 weeks long army training. 
I had no problem with Azure Kinect Body Tracking Viewer. I also rechecked my paths and there were no problems.
However, the same error is still occurring. 

Thank you for your time.

frankkim1108, 2021-02-23:
Luckily, I managed to solve the problem.
I found out that `cannot locate onnxruntime.dll error` had something to with Windows. 
Therefore, I updated my Windows to the newest version and degraded the AzureKinectSDK to v10.0.
 After I tried those solutions, It worked just fine.

closed 2021-02-23 by frankkim1108

ibaiGorordo, 2021-02-26:
Sorry for the delay in the response, I am glad you were able to figure it out.