ibaiGorordo/pyKinectAzure #18: Audio support

LexQzim, 2021-02-22:
Hey there,

is there a way to capture the soundtrack from the Kinect?
In this sense, is it also possible to activate or deactivate corresponding sensors, such as the microphone?
If not, is this possibly planned for the future?

Nice work btw.
Thank you.

ibaiGorordo, 2021-02-26:
Hi,

To be honest, there are other features that have more piority for me than the audio.

Also, it seems that the microphones are not supported in the SDK, so you might be able to use a python library for audio devices. Here is more info, it is a bit old, so things might have changed:

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/536

LexQzim, 2021-03-23:
Ah ok, thanks. Didn't know that.

closed 2021-03-23 by LexQzim