ibaiGorordo/pyKinectAzure #20: Body Tracker Initialization Failed due to "onnxruntime.dll"

VineetVS, 2021-04-16:
I am using your library with with Azure Kinect DK for Pose Estimation. On of the my laptop - it works perfectly.

When I replicated the development environment on another laptop (with the same hardware specification as the first laptop), maintaining the exact same version of NVidia Drivers, CUDA, cuDNN, Azure Kinect SDK and Azure Kinect Body Tracking SDK; I get exception in Body Tracker Initialization due to "onnxruntime.dll"

Below is the log -

> 2021-04-16 13:11:46.1972431 [E:onnxruntime:, inference_session.cc:1268 onnxruntime::InferenceSession::Initialize::<lambda_d0b43ece64526f87430c2f161129335c>::operator ()] Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FF8634BF2AC: (caller: 00007FF86350EE66) Exception(3) tid(4030) 80070057 The parameter is incorrect.

>[2021-04-16 13:11:46.222] [error] [t=16432] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (185): Create(). Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FF8634BF2AC: (caller: 00007FF86350EE66) Exception(3) tid(4030) 80070057 The parameter is incorrect.

>[2021-04-16 13:11:46.222] [error] [t=16432] [K4ABT] ..\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
Body tracker initialization failed!
Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FF8634BF2AC: (caller: 00007FF86350EE66) Exception(3) tid(4030) 80070057 The parameter is incorrect.

>  File "D:\MY_PROJECT_PATH\MY_PYTHON_SCRIPT.py", line 1383, in <module>
    pyK4A.bodyTracker_start(bodyTrackingModulePath)
  File "../pyKinectAzure\pyKinectAzure.py", line 34, in bodyTracker_start
    self.body_tracker = kinectBodyTracker(bodyTrackerModulePath,  depthSensorCalibration)
  File "../pyKinectAzure\kinectBodyTracker.py", line 26, in __init__
    self.initializeTracker()
  File "../pyKinectAzure\kinectBodyTracker.py", line 73, in initializeTracker
    _k4abt.VERIFY(self.k4abt.k4abt_tracker_create(self.sensor_calibration, self.tracker_config, self.tracker_handle), "Body tracker initialization failed!")
  File "../pyKinectAzure\_k4abt.py", line 107, in VERIFY
    traceback.print_stack()

>Process finished with exit code 1

Can you please suggests a solution that I can try to fix this.

I have double checked all the version of the dependencies and everything is matching to the dependencies on the 1st laptop on which this Library is working flawlessly.

Thanks for the help.

ibaiGorordo, 2021-04-18:
Hi, 

I haven't seen that error before, I am not sure what the problem is. Does the Kinect body tracking demo from Microsoft work in that computer?

It might be a problem with onnxruntime, it seems that the same error happens also with other models: https://github.com/microsoft/onnxruntime/issues/2573

VineetVS, 2021-04-19:
Hi,

I manage to fix this issue. The problem was with the version of Azure Kinect Body Tracking SDK. 

So, I tracked the DLL dependencies using 'Process Explorer' in Windows. I found that the DLLs in the Body Tracking SDK in the path

 **'C:\Program Files\Azure Kinect Body Tracking SDK\sdk\windows-desktop\amd64\release\bin'**

were different in the 2 laptops (due to the different version of the Azure Body Tracking SDK). 

In the 1st Laptop on which the Library and code was working - had Azure Body Tracking SDK version 1.0.1 and on the 2nd Laptop on which I got this error had Azure Body Tracking SDK version 1.1.0

So, there seems to be some change in the DLLs going from version 1.0.1 to 1.1.0. I also found more number of DLLs in version 1.1.0. So, it seems the initialization of some DLL which is used by your library is changed in Body Tracking SDK version 1.1.0

Your library works perfectly with Azure Body Tracking SDK version 1.0.1. 

So, I request to please check the above findings and if you find the above mentioned findings correct; please add the Azure Body Tracking SDK version 1.0.1 to Prerequisites.

Thanks.

fractalfantasy, 2021-04-29:
@VineetVS  having a similar issue, did you manage get this working with Body Tracking SDK v1.1.0 or do you have to downgrade to the previous sdk?

VineetVS, 2021-04-30:
@fractalfantasy Please try with the Body Tracking SDK v1.0.1. It seems like there are changes in the DLLs in the Body Tracking SDK v1.1.0. 

So, please downgrade to **Body Tracking SDK v1.0.1** and  please do share your findings.

ibaiGorordo, 2021-04-30:
I will also try to give it a test to the new version during the weekend to check the problem.

ibaiGorordo, 2021-04-30:
The problem is with the new DirectML (Windows only) GPU mode. I was able to make it run by forcing it to CUDA mode. Not sure what the problem is, but for now it should work (there might be some warning about a missing cuda kernel but it should work anyway).

ibaiGorordo, 2021-04-30:
I fixed the DirectML support problem, it should work fine now.

fractalfantasy, 2021-04-30:
awesome!

closed 2021-05-16 by ibaiGorordo