ibaiGorordo/pyKinectAzure #23: Body tracker initialization failed! - due to CUDA Out of Memory Error

VineetVS, 2021-05-20:
Hello:

I am using the Body Tracking SDK in a Flask API to perform Pose Estimation on selected body parts. 

For Example, with FLASK API_1, I am performing tracking of the Hands, with FLASK API_2, I am performing tracking of the head etc. So, I have created multiple APIs and have return call back functions which do body tracking for that specific API call.

Now, I have used code from the example > exampleBodySegmentation.py.

I am doing the initialization of the pyK4A at each API call function using 

`
 pyK4A = pyKinectAzure(modulePath))
`

 and then I am releasing/disposing the instance before exiting the API callback function using 

`
       pyK4A.image_release(depth_image_handle)			
       pyK4A.image_release(pyK4A.body_tracker.segmented_body_img)
        pyK4A.device_stop_cameras()
	pyK4A.device_close()
`

I am using NVidia RTX 2060 GPU and I have observed that with each API Call approx 1 GB VRAM of the GPU is used. So, if I call the API 3 times, on each API Call approx 1 GB VRAM is consumed and it adds up to around 3.5 GB. Due to this - if I make 6 API calls, all the GPU Memory gets consumed (approx 5.2 GB) and then we get the below mentioned error -

`
        
        Device configuration:
        color_format: 0
        (0:JPG, 1:NV12, 2:YUY2, 3:BGRA32)

        color_resolution: 1
        (0:OFF, 1:720p, 2:1080p, 3:1440p, 4:1536p, 5:2160p, 6:3072p)

        depth_mode: 2
        (0:OFF, 1:NFOV_2X2BINNED, 2:NFOV_UNBINNED,3:WFOV_2X2BINNED, 4:WFOV_UNBINNED, 5:Passive IR)

        camera_fps: 2
        (0:5 FPS, 1:15 FPS, 2:30 FPS)

        synchronized_images_only: True
        (True of False). Drop images if the color and depth are not synchronized

        depth_delay_off_color_usec: 0 ms.
        Delay between the color image and the depth image

        wired_sync_mode: 0
        (0:Standalone mode, 1:Master mode, 2:Subordinate mode)

        subordinate_delay_off_master_usec: 0 ms.
        The external synchronization timing.

        disable_streaming_indicator: False
        (True or False). Streaming indicator automatically turns on when the color or depth camera's are in use.

        2021-05-17 11:53:17.9269568 [E:onnxruntime:, inference_session.cc:626 onnxruntime::InferenceSession::Initialize] Exception during initialization: C:\agent\_work\11\s\onnxruntime\core\providers\cuda\cuda_call.cc:97 onnxruntime::CudaCall C:\agent\_work\11\s\onnxruntime\core\providers\cuda\cuda_call.cc:91 onnxruntime::CudaCall CUDA failure 2: out of memory ; GPU=0 ; hostname=XYZ ; expr=cudaMalloc((void**)&p, size);

        Exception during initialization: C:\agent\_work\11\s\onnxruntime\core\providers\cuda\cuda_call.cc:97 onnxruntime::CudaCall C:\agent\_work\11\s\onnxruntime\core\providers\cuda\cuda_call.cc:91 onnxruntime::CudaCall CUDA failure 2: out of memory ; GPU=0 ; hostname=NIBLP746 ; expr=cudaMalloc((void**)&p, size);

        [2021-05-17 11:53:17.953] [error] [t=12268] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (157): Create(). Exception during initialization: C:\agent\_work\11\s\onnxruntime\core\providers\cuda\cuda_call.cc:97 onnxruntime::CudaCall C:\agent\_work\11\s\onnxruntime\core\providers\cuda\cuda_call.cc:91 onnxruntime::CudaCall CUDA failure 2: out of memory ; GPU=0 ; hostname=XYZ ; expr=cudaMalloc((void**)&p, size);
        
        [2021-05-17 11:53:17.953] [error] [t=12268] [K4ABT] D:\a\1\s\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()

         Body tracker initialization failed!

`

After performing some diagnostics by using breakpoints and debugger, as well as monitoring the GPU usage in the Task Manager and with the command 'nvidia-smi'; I was able to conclude that - may be multiple body tracking trackers/instances are being created at each API calls and they are not getting released/disposed. Hence the GPU memory is exhausting after 5 to 6 API calls. Additionally, if we terminate the Flask Server, the GPU Memory gets Flushed out. So, my hypothesis seems reasonable.

I did some more research and I found a link where in people faced similar issue. Please check the last comment by YZR821 on this link -
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/576

**I have tried to delete pyK4A instance and explicitly call the Garbage Collector before exiting the function; but that does not help.**

Additionally, I experiment with the sample example "exampleBodySegmentation.py" by running it 5 times and observed the GPU Memory usage in the Task Manager; but since the Script Terminate when we press ESC Key, I was not able to confirm if some tracker is still utilizing the GPU Memory.

So, can you please check and let me know - how can we properly dispose/release the GPU memory once we are done with the Body Tracking SDK. 

Thanks .

VineetVS, 2021-06-17:
I found a work around to above issue by calling the Body Tracking as Sub Process. This way when the Script called as Sub Process terminate, it will dispose all the Tracker Objects and the GPU Memory is released.

ibaiGorordo, 2021-06-19:
Hi, 
I am sorry, I forgot to respond your initial comment. To be honest, I do not understand completelly about your problem, but if you want to share how you did it or a pull request, there might be someone with the same problem.