ibaiGorordo/pyKinectAzure #27: Body tracker initialization failed!

Interrupt0r, 2021-06-23:
Hello

I am trying to run your demos. Everything is working except of 
exampleBodySegmentation
exampleBodyTracking
exampleBodyTrackingColorCamera

I get this error:

Body tracker initialization failed!
  File "<string>", line 1, in <module>
  File "C:\Users\Interrupt\AppData\Local\Programs\Python\Python37\lib\idlelib\run.py", line 155, in main
    ret = method(*args, **kwargs)
  File "C:\Users\Interrupt\AppData\Local\Programs\Python\Python37\lib\idlelib\run.py", line 548, in runcode
    exec(code, self.locals)
  File "C:\Users\Interrupt\pyKinectAzure\examples\exampleBodyTrackingColorCamera.py", line 35, in <module>
    pyK4A.bodyTracker_start(bodyTrackingModulePath)
  File "../pyKinectAzure\pyKinectAzure.py", line 49, in bodyTracker_start
    self.body_tracker = kinectBodyTracker(bodyTrackerModulePath,  depthSensorCalibration, modelType)
  File "../pyKinectAzure\kinectBodyTracker.py", line 41, in __init__
    self.initializeTracker()
  File "../pyKinectAzure\kinectBodyTracker.py", line 103, in initializeTracker
    _k4abt.VERIFY(self.k4abt.k4abt_tracker_create(self.sensor_calibration, self.tracker_config, self.tracker_handle), "Body tracker initialization failed!")
  File "../pyKinectAzure\_k4abt.py", line 117, in VERIFY
    traceback.print_stack()

My aim is to print the joint coordinates. Maybe you can help me?

(I've installed Azure Kinect SDK v1.4.1 and Azure Kinect Body Tracking SDK 1.1.0  + made the PATH entrys)

ibaiGorordo, 2021-06-23:
Hi, 

I don't see any clue about why there is an error. Does the official "Azure Kinect Body Tracking Viewer" run correctly in your computer?

Also, does the lite version "exampleBodyTrackingLiteModel.py" work correctly?

Interrupt0r, 2021-06-24:
Hello,
thank you for your answer. No Body Tracking *.py is working. Only RGB, IR Depth tests work correctly. k4aviewer.exe is working, but how can I check  Azure Kinect Body Tracking SDK? The exe in tools (k4abt_simple_3d_viewer.exe) doesn't seem to be a 32Bit / 64Bit application? I'm sure I installed everything correctly. My hardware is 3700X AMD CPU and 5700 XT AMD GPU. Do I need a graphics card with CUDA?

ibaiGorordo, 2021-06-24:
Not sure what might be the problem with your "k4abt_simple_3d_viewer.exe", I have tested in my laptop without GPU and it works.

However, I am getting the same error as you when I run my body tracking python programs. Do you also get something like this before the error you posted?
```
[2021-06-25 08:41:37.491] [error] [t=9344] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (500): VerifyDependencies(). Cannot locate cudart64_110.dll; Please install CUDA 11.1 at https://developer.nvidia.com/cuda-11.1.0-download-archive
[2021-06-25 08:41:37.492] [error] [t=9344] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (154): Create(). At least one dependent library is missing!
[2021-06-25 08:41:37.493] [error] [t=9344] [K4ABT] ..\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
```
So, it seems to be possible to run the body tracker without an Nvidia GPU, but I am not sure which of the libraries in the tools folder I need to load to make the body tracker running. Even setting the "processing_mode" in the body tracker configuration to K4ABT_TRACKER_PROCESSING_MODE_CPU returns the same error. 

Even without the error, I am not sure if you will be able to use the AMD GPU to accelerate the body tracking, you should probably search or ask Microsoft about it.

Interrupt0r, 2021-06-27:
Sorry, you are right. k4abt_simple_3d_viewer.exe is working indeed. But how can I test the body tracking feature?

How did you manage to start this "demo":
https://github.com/ibaiGorordo/pyKinectAzure/blob/master/doc/images/outputImage.jpg?raw=true

what specifically did you run?
Are you abled to print out coordinates of any join?

ibaiGorordo, 2021-06-28:
I am glad that it worked, not sure what might have been the problem...

To print you can do somehting like this:
```
bodies = pyK4A.body_tracker.bodiesNow
for body in bodies:
    pyK4A.body_tracker.printBodyPosition(body)
```

after doing `pyK4A.bodyTracker_update()`. You can check the `printBodyPosition` function to see how it is done:
https://github.com/ibaiGorordo/pyKinectAzure/blob/5567de5004186350e735c224ca5f1cf6d582e698/pyKinectAzure/kinectBodyTracker.py#L65 

I am working now on refactoring the code to make these things easier, but not sure when I will be able to get it ready.

Spunky78, 2021-07-19:
Thx a lot, it worked perfectly.

Here are coordinates of the right hand:

x = body.skeleton.joints[_k4abt.K4ABT_JOINT_HAND_RIGHT].position.v[0]
y = body.skeleton.joints[_k4abt.K4ABT_JOINT_HAND_RIGHT].position.v[1]
z = body.skeleton.joints[_k4abt.K4ABT_JOINT_HAND_RIGHT].position.v[2]

here is a table with all joints:
https://docs.microsoft.com/en-us/azure/kinect-dk/body-joints

closed 2021-10-17 by ibaiGorordo