ibaiGorordo/pyKinectAzure #28: Problems with multiprocessing

Chen3110, 2021-07-28:
I am using this module, it works fine for me. Thanks for your great work.

However, it seems that the `pyK4A.device_start_cameras()` doesn't work in sub process or sub thread, while using it in main process will cause a block which is not acceptable.

Could you please give me some advice about this? Or are there any other solution works with multi-process?

Thanks!

ibaiGorordo, 2021-07-29:
Hi, 

I have not tested with multiprocessing. Not sure when I will be able to work on something like that. In the meantime, you can check this other repository where they seem to be able to do what you want:

https://github.com/etiennedub/pyk4a/issues/79

Chen3110, 2021-07-31:
Hi, 

Thanks for your reply. I think the reason for this issue is that the image handle-pointer send to the sub-thread will be updated in the main thread. So the handle will be ineffective. I would be grateful if you have some solutions for the problem.

Thanks!

closed 2021-12-24 by ibaiGorordo

JJLimmm, 2023-02-07:
Hi, i am still unable to run the initialization and RGB/depth camera frame retrieval in a subprocess using Multiprocessing. Is there a fix for this?

@Chen3110 @ibaiGorordo

mroglic, 2023-09-22:
I tried writing version with threads but it wasn't working (I got some errors like - "Invalid k4a_capture_t...") , but doing it all on the main thread worked:

```python
import sys
import cv2

sys.path.insert(1, '../')
import pykinect_azure as pykinect

devices = []

def get_device_capture(device_index, output_queue):
    capture = devices[device_index].update()
    ret, color_image = capture.get_color_image()
    if ret:
        output_queue[device_index] = color_image

if __name__ == "__main__":

    pykinect.initialize_libraries()

    output_queue = [None] * 2 
    
    device_config = pykinect.default_configuration
    device_config.color_resolution = pykinect.K4A_COLOR_RESOLUTION_720P
    device_config.depth_mode = pykinect.K4A_DEPTH_MODE_OFF
 
    device0 = pykinect.start_device(config=device_config, device_index=0)
    device1 = pykinect.start_device(config=device_config, device_index=1) 
    
    devices.append(device0)
    devices.append(device1) 

    window_name = 'Color Image'
    cv2.namedWindow(window_name, cv2.WINDOW_AUTOSIZE)
 
    while True:

        get_device_capture(0, output_queue)
        get_device_capture(1, output_queue)
   
        combined_image = cv2.vconcat([output_queue[0], output_queue[1]])
        cv2.imshow('Color Image', combined_image) 

        if cv2.waitKey(1) == ord('q'):            
            break  
    
    cv2.destroyAllWindows()`