ibaiGorordo/pyKinectAzure #30: Using code causes memory leaks.

euijaebilly, 2021-09-01:
`
def reloading_images(self):
        memory_usage("brfore Threading loop")
        while True:
            memory_usage("screen Threading")
            img1 = pyAzKinct.get_combine_images(self.camera1)  # get image1 from camera1
            memory_usage("check1-1")
            crop_img1 = img1[236:844,656:1264]
            memory_usage("check1-2")
            crop_img1 = cv.cvtColor(crop_img1, cv.COLOR_BGR2RGB)  # convert BGR Channel of image1 to RGB
            memory_usage("check1-3")
            data1 = QByteArray(crop_img1.tostring())
            memory_usage("check1-4")
            qImg1 = QImage(data1, crop_img1.shape[1], crop_img1.shape[0], QImage.Format_RGB888)  # transform converting image1 in RGB 8bit pixel image
            memory_usage("check1-5")
            pImg1 = QPixmap.fromImage(qImg1)
            memory_usage("check1-6")
            self.label1.setPixmap(pImg1) # apply image
            memory_usage("check1-7")
            self.label1.update()  # refreshing image
            memory_usage("check1-8")

            img2 = pyAzKinct.get_combine_images(self.camera2)
            memory_usage("check2-1")
            crop_img2 = img2[236:844,656:1264]
            memory_usage("check2-2")
            crop_img2 = cv.cvtColor(crop_img2, cv.COLOR_BGR2RGB)
            memory_usage("check2-3")
            data2 = QByteArray(crop_img2.tostring())
            memory_usage("check2-4")
            qImg2 = QImage(data2, crop_img2.shape[1], crop_img2.shape[0], QImage.Format_RGB888)
            memory_usage("check2-5")
            pImg2 = QPixmap.fromImage(qImg2)
            memory_usage("check2-6")
            self.label2.setPixmap(pImg2)
            memory_usage("check2-7")
            self.label2.update()
            memory_usage("check2-8")

            #self.log_box.moveCursor(QTextCursor.End)
            memory_usage("check3")
`

`def get_images(camera):
    # Get capture
    memory_usage("get_images 1")
    camera.device_get_capture()

    # Get the depth image from the capture
    depth_image_handle = camera.capture_get_depth_image()

    # Get the color image from the capture
    color_image_handle = camera.capture_get_color_image()

    memory_usage("get_images handle")
    # Check the image has been read correctly
    if depth_image_handle and color_image_handle:
        # Read and convert the image data to numpy array:
        color_image = camera.image_convert_to_numpy(color_image_handle)
        memory_usage("get_images color")
        # Transform the depth image to the color format
        transformed_depth_image = cv2.normalize(camera.transform_depth_to_color(depth_image_handle, color_image_handle), None, 0, 255, cv2.NORM_MINMAX, dtype=cv2.CV_8U)
        memory_usage("get_images depth")
        # Convert depth image (mm) to color, the range needs to be reduced down to the range (0,255)
        transformed_depth_color_image = cv2.applyColorMap(np.round(transformed_depth_image / 30).astype(np.uint8),
                                                          cv2.COLORMAP_JET)
        memory_usage("get_images trans")
        # Add the depth image over the color image:
        combined_image = cv2.addWeighted(color_image[:, :, :3], 0.7, transformed_depth_color_image, 0.3, 0)

[... 173 lines cut ...]
[get_images color] memory usage:  839.18750 MB
[get_images depth] memory usage:  856.38281 MB
[get_images trans] memory usage:  862.32031 MB
[get_images combine] memory usage:  866.24609 MB
[check2-1] memory usage:  852.39844 MB
[check2-2] memory usage:  851.33984 MB
[check2-3] memory usage:  852.39844 MB
[check2-4] memory usage:  853.45703 MB
[check2-5] memory usage:  852.39844 MB
[check2-6] memory usage:  853.81250 MB
[check2-7] memory usage:  852.39844 MB
[check2-8] memory usage:  852.39844 MB
[check3] memory usage:  852.39844 MB
[screen Threading] memory usage:  852.39844 MB
[get_images 1] memory usage:  852.39844 MB
[get_images handle] memory usage:  852.39844 MB
[get_images color] memory usage:  852.39844 MB
`

I created a thread using this function for update frames.
but it makes leaking massive memory(almost 30MB), every each frames.

how can i solve this problem?

also I'm sorry for my poor English.

ibaiGorordo, 2021-09-02:
Hi,

I think it might be related with releasing the images and captures. I will have to check it.

How are you checking the memory leak? So that I can compare the results.

Thank you,
Ibai

euijaebilly, 2021-09-02:
[pre-start] memory usage:  128.82422 MB
[screen Threading] memory usage:  128.87891 MBstarted..

[get_images 1] memory usage:  140.22656 MB
[get_images handle] memory usage:  162.60156 MB
[get_images color] memory usage:  162.66406 MB
[get_images depth] memory usage:  217.75391 MB
[get_images trans] memory usage:  241.04688 MB
[get_images combine] memory usage:  247.01562 MB
[check1-1] memory usage:  239.10156 MB
[check1-2] memory usage:  239.10156 MB
[check1-3] memory usage:  240.16016 MB
[check1-4] memory usage:  241.22656 MB
[check1-5] memory usage:  241.23047 MB
[check1-6] memory usage:  242.66016 MB
[check1-7] memory usage:  242.66016 MB
[check1-8] memory usage:  242.66406 MB
[screen Threading] memory usage:  242.66406 MB
[get_images 1] memory usage:  242.66406 MB
[get_images handle] memory usage:  242.66406 MB
[get_images color] memory usage:  242.66406 MB
[get_images depth] memory usage:  288.49609 MB
[get_images trans] memory usage:  295.50000 MB
[get_images combine] memory usage:  301.43750 MB
[check1-1] memory usage:  287.58984 MB
[check1-2] memory usage:  286.53125 MB
[check1-3] memory usage:  287.58984 MB
[check1-4] memory usage:  288.64844 MB
[check1-5] memory usage:  287.58984 MB
[check1-6] memory usage:  289.00391 MB
[check1-7] memory usage:  287.58984 MB
[check1-8] memory usage:  287.58984 MB
[screen Threading] memory usage:  287.58984 MB
[get_images 1] memory usage:  287.58984 MB
[get_images handle] memory usage:  287.58984 MB
[get_images color] memory usage:  287.58984 MB
[get_images depth] memory usage:  306.61719 MB
[get_images trans] memory usage:  311.58594 MB
[get_images combine] memory usage:  317.51953 MB
[check1-1] memory usage:  303.67188 MB
[check1-2] memory usage:  302.61328 MB
[check1-3] memory usage:  303.67188 MB
[check1-4] memory usage:  304.73047 MB
[check1-5] memory usage:  303.67188 MB
[check1-6] memory usage:  305.08594 MB
[check1-7] memory usage:  303.67188 MB
[check1-8] memory usage:  303.67188 MB
[screen Threading] memory usage:  303.67188 MB
[get_images 1] memory usage:  303.67188 MB
[get_images handle] memory usage:  303.67188 MB
[get_images color] memory usage:  303.67188 MB
[get_images depth] memory usage:  324.22656 MB
[get_images trans] memory usage:  329.18359 MB
[get_images combine] memory usage:  335.11719 MB
[check1-1] memory usage:  321.26953 MB
[check1-2] memory usage:  320.21094 MB
[check1-3] memory usage:  321.26953 MB
[check1-4] memory usage:  322.32812 MB
[check1-5] memory usage:  321.26953 MB
[check1-6] memory usage:  322.68359 MB
[check1-7] memory usage:  321.26953 MB
[check1-8] memory usage:  321.26953 MB
[screen Threading] memory usage:  321.26953 MB
[get_images 1] memory usage:  321.26953 MB
[get_images handle] memory usage:  321.26953 MB
[get_images color] memory usage:  321.26953 MB
[get_images depth] memory usage:  340.57812 MB
[get_images trans] memory usage:  346.57031 MB
[get_images combine] memory usage:  353.50781 MB
[check1-1] memory usage:  339.66016 MB
[... 267 lines cut ...]
[get_images trans] memory usage:  586.51172 MB
[get_images combine] memory usage:  592.44531 MB
[check1-1] memory usage:  578.59766 MB
[check1-2] memory usage:  577.53906 MB
[check1-3] memory usage:  578.59766 MB
[check1-4] memory usage:  579.65625 MB
[check1-5] memory usage:  578.59766 MB
[check1-6] memory usage:  580.01172 MB
[check1-7] memory usage:  578.59766 MB
[check1-8] memory usage:  578.59766 MB
[screen Threading] memory usage:  578.59766 MB
[get_images 1] memory usage:  578.59766 MB
[get_images handle] memory usage:  578.59766 MB
[get_images color] memory usage:  578.59766 MB
[get_images depth] memory usage:  592.60156 MB
[get_images trans] memory usage:  598.03906 MB
[get_images combine] memory usage:  603.97266 MB
[check1-1] memory usage:  590.12500 MB
[check1-2] memory usage:  589.06641 MB
[check1-3] memory usage:  590.12500 MB
[check1-4] memory usage:  591.18359 MB
[check1-5] memory usage:  590.12500 MB
[check1-6] memory usage:  591.53906 MB
[check1-7] memory usage:  590.12500 MB
[check1-8] memory usage:  590.12500 MB

euijaebilly, 2021-09-02:
@ibaiGorordo and the test code in zip, has a method for checking memory usage, called 'memory_usage'

euijaebilly, 2021-09-02:
> I have not tested it, but looking at your code, I would say that the problem is that you removed the **image_release** and **capture_release** functions. Check the original example for reference, and let me know if that fixes the problem.

i edited code files in zip.
`        camera.image_release(depth_image_handle)
        camera.image_release(color_image_handle)

    camera.capture_release()`

add this release code after get_images
and...yes.... it's still leaking....

ibaiGorordo, 2021-09-02:
Yes, I have been doing some testing and the memory leak only happens when running in a thread. I have tried to move the start of the device inside the run function, but it is still leaking.
https://github.com/etiennedub/pyk4a/issues/79

I am not sure what the problem is. In the meantime, this repository might be better for what you are looking for, check the transformation example:
https://github.com/etiennedub/pyk4a

euijaebilly, 2021-09-02:
thanks for your recommand!

closed 2021-09-10 by euijaebilly