ibaiGorordo/pyKinectAzure #31: How to read a record?

victorc365, 2021-09-15:
Hello I have a question, how to read data from the video record? Because I check the samples and didn't see a sample of offline processing. Thank you for your help.

ibaiGorordo, 2021-09-15:
Right now it only records the video. I almost have the rest ready, but I have not polished it enough to publish. I am not sure when I will be able to work on it. So, probably your best option now is to use this other repository:
https://github.com/etiennedub/pyk4a/blob/master/example/playback.py

victorc365, 2021-10-04:
Thank you very much. And an additional question: Is it possible using capture returned like in the share sample as input for body tracking in this library? 

Thank you very much for your response.

ibaiGorordo, 2021-10-04:
Do you mean to perform the body tracking with the recorded sample? I have been a bit inactive, but that is one of the things I want to try to implement soon (probably a couple of weeks).

ibaiGorordo, 2021-10-17:
As mentioned in #35:
"I have uploaded a major refactor that was long overdue. I have added an example "examplePlaybackBodyTracker.py" to perform body tracking in a recording that did not have body tracking enabled."