ibaiGorordo/pyKinectAzure #37: A memory leak problem on the new refactor version.

Lukikay, 2021-11-02:
Hi, thanks for updating this refactor version. The playback feature with body tracking works well for me.

But I think I got a memory leaking problem. On my computer, all example codes would cause a memory leak.
For the `exampleColorImage.py` I manually added the `capture.reset()` method to solve the problem. However, I find it is a bit tricky to release a `body_frame`, I need also to create a new `Frame` after each release, which significantly drops the performance.

Can you help me? And may I know your computer OS and python version? I'm afraid this may be just because of my legacy version of python (I'm working on Windows 10 x64 with python 3.7.7).

cesar-pz, 2021-11-08:
Hello, I had the same issue just last week and I found that there are two different release functions. There is one for normal captures and one for body frame captures. Check out [capture release](https://github.com/ibaiGorordo/pyKinectAzure/blob/97bbd0015ffad7d003ae04c0f46062d0fb7f2c28/pykinect_azure/k4a/_k4a.py#L76) and [body frame release](https://github.com/ibaiGorordo/pyKinectAzure/blob/97bbd0015ffad7d003ae04c0f46062d0fb7f2c28/pykinect_azure/k4abt/_k4abt.py#L84). Be sure to release captures and body frames when you are done using them. Adding these two release calls at the end of each example's while loop fixed the memory leaking for me.

ibaiGorordo, 2021-11-09:
Hi,

Sorry for the delay and thank you both for tracking the issue. I had forgotten to release the native frames/captures. I think the issue should be fixed now.

closed 2021-12-24 by ibaiGorordo

HelloRicky123, 2022-06-03:
Hi, I found that though the current version has a release function for recording, it does not have a release for playback. To see this, just run the examplePlayback.py and the memory will keep growing. I add `capture.release_handle()`, `capture.reset()` and `del capture`, the memory still keeps growing.