ibaiGorordo/pyKinectAzure #39: Linux: wrong library name

mbusch-regis, 2021-11-16:
In utils.py, lines 30 - 32, you have:

    # For non-Arm chips, first check if it is running linux
    if platform.system().lower() == 'linux':
        return 'k4abt.so'

However, this is not the correct file name. The correct name should be:

    # For non-Arm chips, first check if it is running linux
    if platform.system().lower() == 'linux':
        return 'libk4abt.so'

[utils.py.zip](https://github.com/ibaiGorordo/pyKinectAzure/files/7549632/utils.py.zip)

VimalMollyn, 2022-04-01:
Thanks! This worked for me!

alfredplpl, 2022-04-03:
It is nice. You should make a pull request. Or, Do I should so?

ibaiGorordo, 2022-04-03:
Hi @alfredplpl, if you can that would be great. I have been working on other things and have had the time to test the change.

alfredplpl, 2022-04-03:
Thanks @ibaiGorordo ! I will make a pull request for this issue.

closed 2022-04-09 by ibaiGorordo