ibaiGorordo/pyKinectAzure #40: Why I get the depth image with a 2D array?

hlyf-xs, 2021-11-20:
Hello. I run your project on my jeston agx xavier which is a great job! But I confused that the depth image is a 2D array

ibaiGorordo, 2021-11-20:
I am not sure about your question. What data format did you expect?

hlyf-xs, 2021-11-20:
Ok, thanks your reply.  When I used the get_depth_image() function, it returned a 2D array matrix instead of a 3D matrix

ibaiGorordo, 2021-11-20:
I understand now, I think you are referring to the 3D point cloud data. You gat get it using get_pointcloud() instead of get_depth_image().

However, I have not tested it, so it might not work properly.

hlyf-xs, 2021-11-20:
That's right! I want to get the 3D point cloud data to calculate the distance between objects and Kinect DK. But I have annother quetion how can I fuse the RGB image and the 3D point data?

ibaiGorordo, 2021-11-20:
If you check the **get_pointcloud()** function, it has an optional argument called **calibration_type**, by defaults it is set to the depth camera calibration and therefore the point cloud will be relative to the depth camera. But if you pass the **_k4a.K4A_CALIBRATION_TYPE_COLOR** to the **get_pointcloud()** function, it should give you the point cloud with respect to the RGB camera.

you will have to also add `from pykinect_azure.k4a import _k4a` also in the script to get access to _k4a. I have to do change the code to be give a cleaner support, but for now I think it should work (I have not tested it).

hlyf-xs, 2021-11-20:
Thank you very much! I get it!

closed 2021-11-27 by ibaiGorordo