ibaiGorordo/pyKinectAzure #45: Error while using exampleBodySegmentation.py

Avita0207, 2021-12-22:
Hello!
It may be difficult to understand in English from the translation, but I'm sorry.
I'm a college student who is now starting to touch Azure Kinect. I was looking for Azure Kinect to work with python and found your great library.

So, I tried using "exampleBodySegmentation.py", but the following error occurred.
[image]
In order to solve this problem, I saw a case of a similar error like URL in github memory before and decided to see it and put it into practice.
[](https://githubmemory.com/repo/ibaiGorordo/pyKinectAzure/issues/15)
 
Here, it was said that it was solved by updating Windows and lowering the AzureKinectSDK to v10.0, so I thought I would practice it, but I am in trouble because the AzureKinectSDK v10.0 can not be installed.
I would be grateful if you could tell me how to solve it as soon as possible. Thank you.

ibaiGorordo, 2021-12-22:
Hi!

I think the problem might be different. 

- Have you installed the latest Azure Kinect Body Tracking SDK (https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download)?
- Do you have a Nvidia GPU in your system? (The body tracking requires Cuda DNN to be installed).
- Does the example in the Azure Kinect Body Tracking SDK work in your device?

Avita0207, 2022-01-05:
happy New Year!

I'm sorry I couldn't reply because I've been back to my hometown for a long time.

When I checked it, Cuda DNN could not be installed.
I think that the two you have presented are completed.
Does the "Azure Kinect Body Tracking SDK example" mean "k4abt_simple_3d_viewer.exe"? If this is the case, it is working.
Also, when I installed Cuda DNN and tried it, I got an error like this. 

[image]

I know it's an error with no attributes, but I don't know the solution.
I'm new to python, so I'm sorry for the in
convenience, but I'd appreciate it if you could help me.

Thank you.

Avita0207, 2022-01-07:
I'm sorry suddenly.
After installing CudaDNN, I was checking the operation again, but the example of Azure Kinect Body Tracking SDK that should have worked before did not work on the device.
When I try to open it, a window called "3D Visualization" appears, but the image of Azure Kinect is displayed, and the bone of the recognized person should be overlaid, but it does not appear as a white screen.
The question here may be a little different, but if you know the solution, please let me know.

Thank you.

ibaiGorordo, 2022-01-21:
Sorry for the delay. There was a issue in the code. The example should work well now.

Avita0207, 2022-02-01:
Thank you!

Thanks to you, you can now move the sample!

Thank you very much!

closed 2022-02-06 by ibaiGorordo