ibaiGorordo/pyKinectAzure #48: Is it possible to display the coordinates of the joint as a value?

Avita0207, 2022-02-01:
Hello!

Thank you very much for the sample!

I am currently using this library to write a program that displays the coordinates of joints, and I hope to be able to do various things using those coordinates.

The question is, is it possible to display joint coordinates as values ​​in this wonderful library you created?
If you can, please tell me how to do it.
Thank you!

ibaiGorordo, 2022-02-06:
Hi, 

Thanks for letting me know. I have fixed the issue for printing the body. It should work now, you can also get the same result this way:
```
for body in body_frame.get_bodies():
    print(body)
```

However, that will print a other information like the confidence and orientation. You can use something like this to print only the position:
```
for body in body_frame.get_bodies():

    print("Body: ")
    for joint in body.joints:
		
        joint_info = f"{joint.name} position: {joint.position.x},{joint.position.y},{joint.position.z}"
        print(joint_info)
```

Avita0207, 2022-02-07:
Thank you!

Thanks to you, I was able to do it safely! Sincerely thank. Thank you very much!

closed 2022-02-07 by Avita0207