ibaiGorordo/pyKinectAzure #49: Memory overflow problem

YoshikiKubotani, 2022-02-06:
Hi, I am trying to run some code on a remote server using ssh.
I think the code cannot access the window information in this case.
So, I tried to run a code without lines corresponding to window visualization (e.g  cv2.namedWindow('Depth Image',cv2.WINDOW_NORMAL)).

But, it caused the memory overflow to my computer with 16GB CPU even if I used the only 40 seconds mkv video recorded using Kinect DK camera.

I got some error messages as follows.

`[2022-02-07 03:48:24.141] [error] [t=9425] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/tewrapper/tewrapper.c (61): transform_engine_start_helper(). Transform engine create and initialize failed with error code: 204.
[2022-02-07 03:48:24.142] [error] [t=9425] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/tewrapper/tewrapper.c (68): teresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in transform_engine_start_helper()
[2022-02-07 03:48:24.142] [error] [t=9425] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/tewrapper/tewrapper.c (86): transform_engine_start_helper(tewrapper) returned failure in transform_engine_thread()
[2022-02-07 03:48:24.142] [error] [t=9204] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/tewrapper/tewrapper.c (313): tewrapper_create(). Transform Engine thread failed to start
[2022-02-07 03:48:24.142] [error] [t=9204] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/transformation.c (637): transformation_context->tewrapper != NULL returned failure in transformation_create()`

Can somebody know how to solve this problem?

YoshikiKubotani, 2022-02-06:
Actually, I wrote my own code based on `exampleBodyTracking.py` 
 and [offline_processor code of Azure-Kinect-Samples library](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/offline_processor), and **confirmed my code worked properly for a few seconds mkv file (even with the same few errors I put above) .**

And I also checked my code properly call `__del__` which I think releases CPU memory by calling the original Azure Kinect library(e.g `k4a_dll`).

YoshikiKubotani, 2022-02-06:
I ran the same code on another server that has 1000GB memory, but it didn't work again.
And it popped out `malloc(): unsorted double linked list corrupted`.

YoshikiKubotani, 2022-02-11:
I am not sure why this happened, but it may be due to the others' running code interrupting with my code running.
I actually have never experienced this kind of error on python, but this repository calls the C++ code under the hood.
So, it might have caused the problem.

My code finally worked, when nobody are using the server.

closed 2022-02-11 by YoshikiKubotani

LinyeLi60, 2022-05-16:
Some problem