ibaiGorordo/pyKinectAzure #54: Running Playback on synced .mkv files.

juansolana, 2022-03-15:
Hi! I tried running the `examplePlaybackBodyTracker.py` file using an mkv file from a synced recording I did using `k4arecorder`. 

I'm getting the following error:
`[2022-03-15 11:59:44.460] [error] [t=24870] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2022-03-15 11:59:44.460] [error] [t=24870] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (389): Invalid argument to image_get_buffer(). image_handle ((nil)) is not a valid handle of type k4a_image_t
[2022-03-15 11:59:44.460] [error] [t=24870] [K4ABT] ../src/TrackerHost/DepthFrameBlobK4A.cpp (13): Initialize(). Get depth buffer from the capture handle failed!
[2022-03-15 11:59:44.460] [error] [t=24870] [K4ABT] ../src/TrackerHost/TrackerHost.cpp (275): EnqueueCapture(). Initialize DepthFrameBlob failed!
Body tracker capture enqueue failed!
  File "examplePlaybackBodyTracker.py", line 32, in <module>
    body_frame = bodyTracker.update(capture=capture)
  File "../pykinect_azure/k4abt/tracker.py", line 36, in update
    self.enqueue_capture(capture.handle(), timeout_in_ms)
  File "../pykinect_azure/k4abt/tracker.py", line 43, in enqueue_capture
    _k4abt.VERIFY(_k4abt.k4abt_tracker_enqueue_capture(self._handle, capture_handle, timeout_in_ms), "Body tracker capture enqueue failed!")
  File "../pykinect_azure/k4abt/_k4abt.py", line 157, in VERIFY
    traceback.print_stack()
`

For some reason when it starts capturing frames from the mkv file it is empty. This only happen with synced mkv files, if I do it with a recording without synching the kinects, it works fine.

I'm just getting familiar with the code, do you ahve any idea how it could be fixed?

Thanks!

ibaiGorordo, 2022-04-09:
I only have one Kinect camera, so I have not tested it. I probably missed some part that is different when using sync cameras, but it is hard to tell since I cannot test it.

LinyeLi60, 2022-05-18:
The problem is that the **playback.isOpened()** will first get next capture and go back to the previous capture, if mkv file is not full 30fps, some capture may lost. So you can replace **while playback.isOpened()** by **while True**. Also check if the fps of depth image is consistent with color image, it will also cause problems.

qiaoxu123, 2022-12-05:
Hi, maybe i am facing the same problem, when i start running the `examplePlayback.py` with the synced .mkv files, it broken with the below message

```
Traceback (most recent call last):
  File "examplePlayback.py", line 15, in <module>
    playback = pykinect.start_playback(video_filename)
  File "E:\Project\pyKinectAzure\examples\..\pykinect_azure\pykinect.py", line 67, in start_playback
    return Playback(filepath)
  File "E:\Project\pyKinectAzure\examples\..\pykinect_azure\k4arecord\playback.py", line 18, in __init__
    self.open(filepath)
  File "E:\Project\pyKinectAzure\examples\..\pykinect_azure\k4arecord\playback.py", line 26, in open
    _k4arecord.VERIFY(_k4arecord.k4a_playback_open(filepath,self._handle),"Failed to open recording!")
  File "E:\Project\pyKinectAzure\examples\..\pykinect_azure\k4arecord\_k4arecord.py", line 83, in k4a_playback_open 
    return _k4a_playback_open(file_path, playback_handle)
ctypes.ArgumentError: argument 1: <class 'TypeError'>: wrong type
[2022-12-05 10:56:41.084] [error] [t=21872] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\include\k4ainternal/matroska_read.h (143): k4a_playback_t_get_context(). Invalid k4a_playback_t 0000000000000000
[2022-12-05 10:56:41.084] [error] [t=21872] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\record\sdk\playback.cpp (658): Invalid argument to k4a_playback_close(). playback_handle (0000000000000000) is not a valid handle of type k4a_playback_t
```
It means that i can't start this playback.  And I get the same result on the Ubuntu 20.04, so it's not caused by the OS.

Now i don't have other mkv files to test, but i guess this problem is caused by the same reason, so are there any ideas how it could be fixed? 

Thanks.

closed 2023-01-06 by ibaiGorordo