ibaiGorordo/pyKinectAzure #55: [BodyTrackingColorCamera] 'Body tracker capture enqueue failed!'

ytop0386, 2022-03-31:
Hi I tried running the `exampleBodyTrackingColorCamera.py`
It did work, and video window was opened, but it shutdown after a while
it occurred below these errors

```
[2022-03-31 11:01:59.335] [error] [t=9820] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2022-03-31 11:01:59.358] [error] [t=9820] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (389): Invalid argument to image_get_buffer(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t
[2022-03-31 11:01:59.359] [error] [t=9820] [K4ABT] ..\src\TrackerHost\DepthFrameBlobK4A.cpp (13): Initialize(). Get depth buffer from the capture handle failed!
[2022-03-31 11:01:59.360] [error] [t=9820] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (275): EnqueueCapture(). Initialize DepthFrameBlob failed!
Body tracker capture enqueue failed!
  File "BodyTrackingColorCamera.py", line 32, in <module>
    body_frame = bodyTracker.update()
  File "..\pykinect_azure\k4abt\tracker.py", line 38, in update
    self.enqueue_capture(Device.capture.handle(), timeout_in_ms)
  File "..\pykinect_azure\k4abt\tracker.py", line 43, in enqueue_capture
    _k4abt.VERIFY(_k4abt.k4abt_tracker_enqueue_capture(self._handle, capture_handle, timeout_in_ms), "Body tracker capture enqueue failed!")
  File "..\pykinect_azure\k4abt\_k4abt.py", line 157, in VERIFY
    traceback.print_stack()
```
when i tried running the `exampleBodyTracking.py`, these error won't happen

Did anyone know how to solve this problem?
thank you lot!

ibaiGorordo, 2022-04-09:
Were you able to solve the issue? Sometimes I have a similar problem if the cable or the USB port is does not support enough USB speed, but not sure if it is the same issue.

HelloRicky123, 2022-05-25:
Hi. I found it may appear if the capture is empty. I use the following code to jump this frame.
```
ret, depth_color_image = capture.get_colored_depth_image()
if not ret:
    continue
```

closed 2023-01-06 by ibaiGorordo

zulkafilabbas, 2024-03-28:
I think this has to do with [https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__device__configuration__t_a8208974f05d89fc1362c6a0900bdef4d.html](url)

See: "https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1416" 

[1]:
```
device_config = pykinect.default_configuration
# device_config.color_resolution = pykinect.K4A_COLOR_RESOLUTION_1080P
# device_config.depth_mode = pykinect.K4A_DEPTH_MODE_WFOV_2X2BINNED
# device = pykinect.start_device(config=device_config)
```
Config [1] was causing (a similar) error:

```
[2024-03-28 08:35:02.147] [error] [t=8488] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2024-03-28 08:35:02.148] [error] [t=8488] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (389): Invalid argument to image_get_buffer(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t
[2024-03-28 08:35:02.148] [error] [t=8488] [K4ABT] ..\src\TrackerHost\DepthFrameBlobK4A.cpp (13): Initialize(). Get depth buffer from the capture handle failed!
[2024-03-28 08:35:02.148] [error] [t=8488] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (274): EnqueueCapture(). Initialize DepthFrameBlob failed!
Body tracker capture enqueue failed!
```

Fixed by modifying Config [1] to [2].

[2]
```
device_config.color_resolution = pykinect.K4A_COLOR_RESOLUTION_1080P
device_config.depth_mode = pykinect.K4A_DEPTH_MODE_NFOV_UNBINNED
device_config.color_format = pykinect.K4A_IMAGE_FORMAT_COLOR_MJPG
device_config.camera_fps = pykinect.K4A_FRAMES_PER_SECOND_30
device_config.synchronized_images_only = True`
```