ibaiGorordo/pyKinectAzure #57: How to save the result of Body Tracking as .json or .skeleton?

110204362, 2022-04-09:
How to save the result of Body Tracking as .json or .skeleton?

ibaiGorordo, 2022-04-09:
I have just added support, you can do:
`json.dumps(body_frame.json())`

I have also added the json() function to body, so something like this is also possible
```
bodies = body_frame.get_bodies()
json.dumps(bodies[0].json())
```

110204362, 2022-04-09:
Thank you very much!

110204362, 2022-04-10:
Thanks for your help !
Excuse me. Could you please tell me how to save the position [x y z] of  joints in a action like the appendix?It really bothered me for a long time.
甄帅
***@***.***

ibaiGorordo, 2022-04-10:
I don't understand your message fully, but I have added a numpy() method () that will return the joint positions like a numpy array. From there, i think you should be able to save it however you like.

Example:
```
bodies = body_frame.get_bodies()
print(bodies[0].numpy())
```

Output:
```
[[  32.12897491  332.56265259  616.41009521]
 [  37.96852493  159.21200562  642.97619629]
 [  39.43162918   19.98685265  660.09326172]
 [  34.05524063 -192.30418396  682.47003174]
 [  61.67406082 -160.24394226  659.37390137]
 [ 179.97952271 -147.86459351  584.41168213]
 [ 237.50953674  107.9463501   512.82244873]
 [  47.29102707   69.35308838  392.95928955]
 [ -13.03019714   57.42243576  320.92556763]
 [ -86.37052917   85.86664581  392.38882446]
 [ -33.30407715   35.25019073  293.20074463]
 [   9.57022762 -154.08050537  699.72607422]
 [ -90.44200134 -118.60158539  773.68292236]
 [-251.91938782  102.75777435  813.94091797]
 [-401.73779297  109.77137756  638.2401123 ]
 [-470.54302979   50.9245224   591.31604004]
 [-521.84375      69.39809418  502.2116394 ]
 [-470.69454956   79.34676361  527.40454102]
 [ 103.20992279  326.62615967  562.04858398]
 [ -18.65794563  630.41009521  343.1796875 ]
 [  59.75688171  993.14758301  405.45117188]
 [   9.51986408 1125.55249023  287.56869507]
 [ -31.96784019  337.91586304  665.43023682]
 [-208.79342651  631.73553467  472.79275513]
 [-122.96312714 1000.97277832  502.74520874]
 [-197.89836121 1116.00061035  403.28289795]
 [  23.93781471 -271.98184204  680.42266846]
 [ -35.08774185 -327.61773682  544.77862549]
 [   3.28371119 -358.12191772  569.83483887]
 [  98.4186554  -334.8977356   647.01531982]
 [ -45.3763237  -355.29214478  586.22406006]
 [ -50.56934357 -336.3182373   713.58001709]]
 ```

closed 2022-05-03 by ibaiGorordo

YongSeongLee25, 2022-06-28:
> I don't understand your message fully, but I have added a numpy() method () that will return the joint positions like a numpy array. From there, i think you should be able to save it however you [...]

hi
Is this data Millimeter value of xyz?

ibaiGorordo, 2022-06-28:
Yes, and here is the list of which joint refers to each row:
https://github.com/ibaiGorordo/pyKinectAzure/blob/538111ccbe1c0c19151ed742e143bf32a0d90978/pykinect_azure/k4abt/_k4abtTypes.py#L25