ibaiGorordo/pyKinectAzure #60: Unable to read multiple cameras data

TitansWhale, 2022-05-08:
I have two kinect cameras and I want to read their data at the same time and display it, but the data I can only read is the same camera. I don't know where is my mistake. My code is as follows.

`
import sys
import cv2

from cfg import config

sys.path.insert(1, '../')
import pykinect_azure as pykinect

def sample_run():
    # Initialize the library, if the library is not found, add the library path as argument
    pykinect.initialize_libraries(track_body=True)

    # Modify camera configuration
    device_config = config.get_kinect_config()
    device_config1 = config.get_kinect_config()
    # print(device_config)

    # Start device
    device = pykinect.start_device(device_index=0, config=device_config)
    device1 = pykinect.start_device(device_index=1, config=device_config1)

    while True:

        # Get capture
        capture = device.update()
        capture1 = device1.update()
        # get rgb
        sucess, img = capture.get_color_image()
        if sucess:
            img = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)

            cv2.imshow("0", img)
            cv2.waitKey(1)
        sucess, img1 = capture1.get_color_image()

        if sucess:
            img1 = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)
            cv2.imshow("1", img)
            cv2.waitKey(1)

if __name__ == "__main__":
    sample_run()
`

TitansWhale, 2022-05-08:
`import sys
import cv2

sys.path.insert(1, '../')
import pykinect_azure as pykinect

def sample_run():
    # Initialize the library, if the library is not found, add the library path as argument
    pykinect.initialize_libraries(track_body=True)

    # Modify camera configuration
    device_config = pykinect.default_configuration
    device_config1 = pykinect.default_configuration
    # print(device_config)

    # Start device
    device = pykinect.start_device(device_index=0, config=device_config)
    device1 = pykinect.start_device(device_index=1, config=device_config1)

    try:
        while True:

            # Get capture
            capture = device.update()

            print(id(capture))

            sucess, img = capture.get_color_image()
            if sucess:
                img = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)

                cv2.imshow("0", img)
                cv2.waitKey(1)

            capture1 = device1.update()
            print(id(capture1))
            sucess, img1 = capture1.get_color_image()

            if sucess:
                img1 = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)
                cv2.imshow("1", img)
                cv2.waitKey(1)
            # get rgb

    finally:
        device.stop_cameras()
        device.stop_cameras()

if __name__ == "__main__":
    sample_run()
`

TitansWhale, 2022-05-11:
thank you for share your code.
I am trying to add multiple cameras feature in pyKinectAzure.

ibaiGorordo, 2022-05-11:
Hi,

Do you get any error? I only have one, so that is why I am not sure about what could be the issue. If you find the solution I will happy to accept a pull request. 

Otherwise, https://github.com/etiennedub/pyk4a or the official python library might work with multiple cameras better.

LinyeLi60, 2022-06-24:
Replace pykinect.default_configuration by Configuration(), don't use pykinect.default_configuration.

TitansWhale, 2022-06-24:
> Replace pykinect.default_configuration by Configuration(), don't use pykinect.default_configuration.

Thanks for your reply, I've solved it and put it in my Repositories.https://github.com/TitansWhale/pyKinectAzure

closed 2022-07-08 by ibaiGorordo