ibaiGorordo/pyKinectAzure #65: AttributeError: 'Tracker' object has no attribute '_handle'

gengauss, 2022-06-01:
Could anyone help me to solve this issue?

JReesW, 2022-09-16:
I got the same error when trying to run exampleBodyTrackingColorCamera.py, and also when trying to call pykinect.start_body_tracker() myself.

I replaced my local path with .....
```
Traceback (most recent call last):
  File ".....\pyKinectAzure-master\examples\exampleBodyTrackingColorCamera.py", line 22, in <module>
    bodyTracker = pykinect.start_body_tracker()
  File ".....\pyKinectAzure-master\examples\..\pykinect_azure\pykinect.py", line 63, in start_body_tracker
    return Tracker(Device.calibration, model_type)
  File ".....\pyKinectAzure-master\examples\..\pykinect_azure\k4abt\tracker.py", line 12, in __init__
    self._handle = self.create(model_type)
  File ".....\pyKinectAzure-master\examples\..\pykinect_azure\k4abt\tracker.py", line 68, in create
    _k4abt.VERIFY(_k4abt.k4abt_tracker_create(self.calibration.handle(), tracker_config, tracker_handle), "Body tracker initialization failed!")
  File ".....\pyKinectAzure-master\examples\..\pykinect_azure\k4abt\_k4abt.py", line 31, in k4abt_tracker_create
    return _k4abt_tracker_create(sensor_calibration, config, tracker_handle)
OSError: exception: access violation reading 0x0000015C8D33C918
Exception ignored in: <function Tracker.__del__ at 0x00000158FA06B1F0>
Traceback (most recent call last):
  File ".....\pyKinectAzure-master\examples\..\pykinect_azure\k4abt\tracker.py", line 17, in __del__
    self.destroy()
  File ".....\pyKinectAzure-master\examples\..\pykinect_azure\k4abt\tracker.py", line 29, in destroy
    if self.is_valid():
  File ".....\pyKinectAzure-master\examples\..\pykinect_azure\k4abt\tracker.py", line 20, in is_valid
    return self._handle
AttributeError: 'Tracker' object has no attribute '_handle'
```

If you need any more information, feel free to ask.

ibaiGorordo, 2022-09-16:
@JReesW in the error it says "Body tracker initialization failed!", so that means that it was not able to find the body tracker library.

Do you have it installed in your computer? If so, does the official example that comes with the the body tracking sdk work in your computer? If it works, you might have to change the library path.

JReesW, 2022-09-16:
@ibaiGorordo I have the body tracker installed at `C:/Program Files/Azure Kinect Body Tracking SDK`. It seems to be loading in fine. (The path defined in `pykinect.py` on line 41 is the exact same path as where my Body Tracking SDK is installed, and the directml.dll file is indeed located there)

The "Body tracker initialization failed!" message is an argument given to _k4abt.py's VERIFY function, meaning that were it to fail verification it would print that message to the console. However, that message doesn't get printed, it just happens to be part of the call stack.

The `OSError: exception: access violation reading 0x...` error gets raised when trying to call `_k4abt_tracker_create(sensor_calibration, config, tracker_handle)` in _k4abt.py on line 31. Everything else up to that point seems to work fine.

JReesW, 2022-09-16:
So I tried running the example files `k4abt_benchmark` and `k4abt_simple_3d_viewer` and I ony got the simple 3d viewer to work when passing the argument `CPU` to it. I have a feeling this might go deeper than errors in the library...

ibaiGorordo, 2022-09-16:
I guess you don't have a Nvidia GPU, right? In that case, you can try to modify this line:
https://github.com/ibaiGorordo/pyKinectAzure/blob/87d722ef5b0ca97b1f595f39d395992a603e56d3/pykinect_azure/pykinect.py#L43

And change it with K4ABT_TRACKER_PROCESSING_MODE_CPU. You will have to also modify the import at the top to this:
```
from pykinect_azure.k4abt._k4abtTypes import k4abt_tracker_default_configuration, K4ABT_TRACKER_PROCESSING_MODE_CPU
```

JReesW, 2022-09-16:
I got it to work!

I didn't only have to change that line inside the `except` clause but I added that line again in the `try` clause of which the `except` is a part. This because it never raised an exception inside the try, so changing the processing mode in the except would have no effect.

It now looks like this in `pykinect.py`:
```
def init_k4abt(module_k4abt_path):

    _k4abt.setup_library(module_k4abt_path)

    try:
        k4abt_tracker_default_configuration.processing_mode = K4ABT_TRACKER_PROCESSING_MODE_CPU
        ctypes.cdll.LoadLibrary("C:/Program Files/Azure Kinect Body Tracking SDK/tools/directml.dll")
    except Exception as e:
        k4abt_tracker_default_configuration.processing_mode  = K4ABT_TRACKER_PROCESSING_MODE_CPU
```

Now to continue on this adventure. Let's hope I don't run into more errors because of inadequate hardware lol.

Thank you very much for helping me, and for creating this wrapper!

ibaiGorordo, 2022-09-16:
Nice! Yeah I should probably do a better cleaning on the library loading.

Hopefully, the rest goes smoothly. I will close this issue, open a new issue if you get any other errors.

closed 2022-09-16 by ibaiGorordo