ibaiGorordo/pyKinectAzure #66: Hi, I found that though the current version has a release function for recording, it does not have a release for playback. To see this, just run the examplePlayback.py and the memory will keep growing. I add `capture.release_handle()`, `capture.reset()` and `del capture`, the memory still keeps growing.

HelloRicky123, 2022-06-03:
Hi, I found that though the current version has a release function for recording, it does not have a release for playback. To see this, just run the examplePlayback.py and the memory will keep growing. I add `capture.release_handle()`, `capture.reset()` and `del capture`, the memory still keeps growing.

_Originally posted by @HelloRicky123 in https://github.com/ibaiGorordo/pyKinectAzure/issues/37#issuecomment-1145815886_

HelloRicky123, 2022-06-03:
Hi, I have found the solution. So I close this issue.
One problem is `while playback.isOpened():`, it will keep increasing the memory. I replace it with `while True:`.
The second problem is no release for playback. I add `if self.is_capture_initialized():self._capture.release_handle()` to the function `get_next_capture` in playback.py.

Thanks a lot for your work.

closed 2022-06-03 by HelloRicky123

angelicaCruz, 2023-01-05:
> Hi, I have found the solution. So I close this issue. One problem is `while playback.isOpened():`, it will keep increasing the memory. I replace it with `while True:`. The second problem is no [...]

hi @HelloRicky123 ! I ran to this issue too and I am trying to use your solution however I am trying to understand the piece you modified. 
Do you put `self._capture.release_handle()` before the `self.capture._handle = capture_handle` or after?
Thank you both for your work!

ibaiGorordo, 2023-01-06:
Sorry for the delay, I just updated the code to fix the issue. The issue @angelicaCruz was that replacing `while playback.isOpened():`. was necessary because of the logic I had to check if the playback was still open.

I think it should work now properly.

angelicaCruz, 2023-01-06:
@ibaiGorordo ah perfect! thank you soo much!