ibaiGorordo/pyKinectAzure #70: get body joint coordinates from raw ir and depth images

abhi5454, 2022-07-28:
I am currently using a python application extract and dump IR and depth images in .raw format. I need to run body tracking on those extracted frames and generate corresponding 3d body joint coordinates. How can i create a custom capture object to use my raw inputs and perform bodytracking on the same to extract bodyjoints .
Input -

Kinect IR image (in .raw format)
Kinect Depth image (in .raw format)
Output
3D coordinates of Bodyjoints

maxythom, 2022-12-16:
I am trying to do a similar thing regarding extracting the body joints, please let me know if you found a solution

ibaiGorordo, 2023-01-06:
It might be possible if you create an image handle from the raw buffer of those images using the `k4a_image_create_from_buffer` function below, and then create a capture with those images to be passed to the body tracker. But that would require a lot of changes. My recommendation is to use the official record method.

https://github.com/ibaiGorordo/pyKinectAzure/blob/fc16de16b40753b8f4aa50c78a2c11bcaecddc4a/pykinect_azure/k4a/_k4a.py#L184-L208

closed 2023-01-06 by ibaiGorordo