ibaiGorordo/pyKinectAzure #71: capture.get_transformed_color_image()

peymoon, 2022-08-09:
I am calling the capture.get_transformed_color_image() on a recorder video (playback) and it returns the following error:
`[error] [t=1092] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (1253): k4a_transformation_color_image_to_depth_camera(). Require color image and transformed color image both have bgra32 format.`

It requires the color image to be K4A_IMAGE_FORMAT_COLOR_BGRA32. However, the color image is K4A_IMAGE_FORMAT_COLOR_MJPG. Simply changing the color image format does not seem to work. Is there anything else that needs to be done before calling the get_transformed_color_image function?

diu-diu-diu, 2022-08-10:
I meet the same problem, how can I use this function `get_transformed_color_image()`  while `MJPG` Color Input

ibaiGorordo, 2023-01-06:
Unfortunately, as you mention, the SDK requires the color image to be BGRA32. So if the recording was taken with MJPG, there is nothing that can be done. You need to change the recording example to have the color format to BGRA32 (check the color-to-depth transformation example).

In theory, it might be possible to get the MJPG, convert it to numpy, convert it to BGRA and the use the `k4a_image_create_from_buffer` function to create a new image from the buffer. But it takes some callback arguments which I am not sure how they need to look like:
https://github.com/ibaiGorordo/pyKinectAzure/blob/fc16de16b40753b8f4aa50c78a2c11bcaecddc4a/pykinect_azure/k4a/_k4a.py#L184-L208

closed 2023-01-06 by ibaiGorordo

miraclejzd, 2023-04-01:
Hi, I also meet the problem. ```k4a_transformation_color_image_to_depth_camera(). Require color image and transformed color image both have bgra32 format.```
The the recording color mode can not be set to BGRA32. Can you give me an example about how to use the function ```k4a_image_create_from_buffer```?