ibaiGorordo/pyKinectAzure #78: _k4abt.VERIFY always fails when trying to run Frame.get_body_skeleton()

JReesW, 2022-09-23:
I made a small example which raises this error for me
```
import pykinect_azure.pykinect as pk

pk.initialize_libraries(track_body=True)
device = pk.start_device()
body_tracker = pk.start_body_tracker()

capture = device.update()
body_frame = body_tracker.update()

ret, skeleton = body_frame.get_body_skeleton()
```

When running this I get the following error:
```
  File ".....\app\minimum_example.py", line 15, in <module>
    ret, skeleton = body_frame.get_body_skeleton()
  File ".....\app\pykinect_azure\k4abt\frame.py", line 53, in get_body_skeleton
    _k4abt.VERIFY(_k4abt.k4abt_frame_get_body_skeleton(self._handle, index, skeleton), "Body tracker get body skeleton failed!")
  File ".....\app\pykinect_azure\k4abt\_k4abt.py", line 157, in VERIFY
    traceback.print_stack()
Body tracker get body skeleton failed!
```

It is entirely possible that I am using it incorrectly, if so, please tell me how it should actually be done.
My end goal is to just get the coordinates of all the joints in a single capture, be it a list or tuple or string or .csv file. Preferably not an image though.
Something like:
```
# jointID, x, y, z, rot_x, rot_y, rot_z, deviceID, certainty
0, 97.2257, -148.314, 1595.26, -0.121209, 0.689724 ,-0.689459, 000808520212, 2
1, 96.4624, -303.488, 1522.69, -0.127357, 0.712497, -0.665898, 000808520212, 2
2, 103.007 ,-427.133, 1464.17, -0.136808, 0.699438, -0.687741, 000808520212, 2
```

Feel free to ask me for more information. Thanks in advance!

ibaiGorordo, 2022-09-23:
My guess is that you are trying to get a skeleton, even if there is no body detected.
Just to make sure, when you run any of the other body-tracking examples, do you get the same error?

What you can do instead is something like this (I haven't tested it so it might need fixes): 
```
import pykinect_azure.pykinect as pk
import numpy as np

pk.initialize_libraries(track_body=True)
device = pk.start_device()
body_tracker = pk.start_body_tracker()

capture = device.update()
body_frame = body_tracker.update()

bodies = body_frame.get_bodies()
body_list = []
for body in bodies:
    body_list.append(body.numpy())

    
```

ibaiGorordo, 2022-09-23:
Also, you can also convert to json in case it is easier to save: https://github.com/ibaiGorordo/pyKinectAzure/issues/57

ibaiGorordo, 2022-09-23:
However, that will only give the position of the joints, you can modify the numpy method of the joint class (https://github.com/ibaiGorordo/pyKinectAzure/blob/0766c3c70622a00bdf43e6875521d6a59dafc7c2/pykinect_azure/k4abt/joint.py#L19) to also save the rotation and confidence

If you do it, please submit a PR

JReesW, 2022-09-23:
I've submitted a PR adding the extra information about orientation and confidence level.

The problem arose because I didn't expect it to crash when nobody was actually standing in front of the camera.
It does work when I do stand in front of it, and I can now get all the information I need.

Just gotta add a try-clause when trying to capture a skeleton, or just use the `.get_bodies()` method, which seems to provide all the information I actually need.

Thanks for your help again!