ibaiGorordo/pyKinectAzure #86: exampleRecording.py error

nliedmey, 2022-12-25:
Hello,
first of all thank you for this great repository! I integrated this repo in my project and tested out some stuff. 
Currently I am struggeling with calling the recording functionality. It works when i run the exampleRecording.py, but calling the method from another class or copying its content into another class always throws error when i start recording.

Anybody has an idea how these error come up?

Here are examplary error stack when calling the function from a Dash server function:

  File "C:\Users\XX\PycharmProjects\pythonProject2\main.py", line 181, in button_start_wrapper_clicked
    examples.exampleRecording.main()
  File "C:\Users\XX\PycharmProjects\pythonProject2\examples\exampleRecording.py", line 19, in main
    device = pykinect.start_device(config=device_config, record=True, record_filepath=video_filename)
  File "C:\Users\XX\PycharmProjects\pythonProject2\pykinect_azure\pykinect.py", line 55, in start_device
    device.start(config, record, record_filepath)
  File "C:\Users\XX\PycharmProjects\pythonProject2\pykinect_azure\k4a\device.py", line 42, in start
    self.record = Record(self._handle, self.configuration.handle(), record_filepath)
  File "C:\Users\XX\PycharmProjects\pythonProject2\pykinect_azure\k4arecord\record.py", line 9, in __init__
    self.create_recording(device_handle, device_configuration, filepath)
  File "C:\Users\XX\PycharmProjects\pythonProject2\pykinect_azure\k4arecord\record.py", line 15, in create_recording
    _k4arecord.VERIFY(_k4arecord.k4a_record_create(filepath.encode('utf-8'), device_handle, device_configuration, self.record_handle),"Failed to create recording!")
  File "C:\Users\XX\PycharmProjects\pythonProject2\pykinect_azure\k4arecord\_k4arecord.py", line 425, in VERIFY
    traceback.print_stack()

[2022-12-25 11:28:27.733] [error] [t=21372] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (53): k4a_device_t_get_context(). Invalid k4a_device_t 0000013ED74D1330
[2022-12-25 11:28:27.733] [error] [t=21372] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (967): Invalid argument to k4a_device_get_version(). device_handle (0000013ED74D1330) is not a valid handle of type k4a_device_t
[2022-12-25 11:28:27.733] [error] [t=21372] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\record\sdk\record.cpp (278): k4a_device_get_version(device, &version_info) returned failure in k4a_record_create()
[2022-12-25 11:28:27.733] [error] [t=21372] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (53): k4a_device_t_get_context(). Invalid k4a_device_t 0000013ED74D1330
[2022-12-25 11:28:27.733] [error] [t=21372] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (959): Invalid argument to k4a_device_get_serialnum(). device_handle (0000013ED74D1330) is not a valid handle of type k4a_device_t
[2022-12-25 11:28:27.733] [error] [t=21372] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\record\sdk\record.cpp (292): k4a_device_get_serialnum(device, serial_number_buffer, &serial_number_buffer_size) returned failure in k4a_record_create()
Failed to create recording!

ibaiGorordo, 2023-01-06:
Without the code, it is hard for me to tell. my guess is that it is probably due to how I implemented the module path "installation", where it is necessary to add the base folder of the repository to the system path `sys.path.insert(1, '../')`, if your main code is somewhere else, make sure that you modify the sys.path insert call to point to the correct position.

Another option, is how you encapsulated the code in the example to a main function. Instead of that, try to take the code in the example and add it inside your class.

nliedmey, 2023-04-11:
Thanks for your comment :)

Sorry can't reproduce currently and not relevant for me anymore. Gonna close this one.