ibaiGorordo/pyKinectAzure #87: Segmentation Fault

angelicaCruz, 2023-01-06:
I don't know if this is an issue on my end but I am running the updated code and I and I get segmentation fault when running body tracker related examples. I can record a video and play it but I cannot process it with the body tracking example anymore. 

Thanks!

angelicaCruz, 2023-01-06:
Not code related

ibaiGorordo, 2023-01-07:
It seems to be related with this: https://github.com/ibaiGorordo/pyKinectAzure/issues/59

The issue is probably with ctypes, because that line you mentioned is just creating a ctypes pointer: 

https://github.com/ibaiGorordo/pyKinectAzure/blob/26eeb55bcf7eed217c92557da7457303dea39027/pykinect_azure/k4abt/_k4abtTypes.py#L7-L11

To get a better idea, can you provide more info:
- what Python version are you using?
- What OS are you using?
- Can you check the ctypes version?
```
import ctypes
print(ctypes.__version__) 
```

ibaiGorordo, 2023-01-07:
The CUDA onnx provider load was missing. I have just added it for Windows (I will check Linux later).

angelicaCruz, 2023-01-11:
this happened due to env setting actually. I missed the tracker part lol. I was closing this but I forgot. Thanks though!! 
Do you happen to know where I can set the depth limit? I am trying to use you wrapper and the tracker to some in-bed pose estimation but I noticed that when I go in bed, the tracker stops working. If I lay on the floor, it works okay-ish but never works in bed situation. It may be that there is no sample in the training where people are laying down or the depth of the bed is calculated badly buy idk. These are speculations I am making.

closed 2023-01-11 by angelicaCruz