ibaiGorordo/pyKinectAzure #88: Is it possible to perform body tracking on an existing .mkv?

benpocklingtonhes, 2023-01-17:
Hello, I have a lot of recordings from the Kinects in .mkv, and we need to perform body tracking on them. Is it possible with this repository to do body tracking on already recorded videos? I can see Microsoft has an "OFFLINE" mode for doing this on their GitHub but unfortunately I don't know C++ and am getting nowhere with it...

ibaiGorordo, 2023-01-17:
Hi, 
Check this example: https://github.com/ibaiGorordo/pyKinectAzure/blob/master/examples/examplePlaybackBodyTracker.py

It might need some tweaks depending on you device configuration. For example, if your recording was in MJPG, you cannot get the color image aligned to the depth map `ret_color, color_image = capture.get_transformed_color_image()`, but you can still get the normal RGB image `ret_color, color_image = capture.get_color_image()`. In that case, comment this line since you cannot combine the color map with the depth map: 
https://github.com/ibaiGorordo/pyKinectAzure/blob/b4200dd1b168a2f3919ca6944a160ebbfc8ad135/examples/examplePlaybackBodyTracker.py#L51

benpocklingtonhes, 2023-01-17:
Oh wow, how did I miss that example haha. Thank you, I will test it out later today

benpocklingtonhes, 2023-01-17:
This is working perfectly, apart from the aligned color map which shouldn't be a problem. Thanks again!

closed 2023-01-17 by benpocklingtonhes