ibaiGorordo/pyKinectAzure #91: IR camera inquiry

DOG-j, 2023-01-25:
It was nice to be able to use all the examples.

When using an IR camera, the contrast is not good as shown in the picture below. What is the problem?
[image]

This phenomenon is the same in all lighting conditions and in all environments.

Thanks

jonas-hurst, 2023-02-08:
The reason for this phenomenon is that pixel values in IR capture are relatively small, yet 16bit arrays are used which have a max value of 65,535. As pixel values are so small, most things appear black.

You must choose a minimum and maximum pixel value range, then stretch all pixel values (e.g. linearly) to fully cover the color depth space you want to use.

For example: Min-Max range is 100-5000.
If you want to work with 8-bit color depth, stretch it to 0-255.
If you want to work with 16-bit color depth, stretch it to 0-65,535

ibaiGorordo, 2023-02-10:
Yeah, this is an old issue I have. I haven't been able to draw the IR image to be similar to the visualizer yet.

In theory, it uses these limits (0-1000) as the range: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/2feb3425259bf803749065bb6d628c6c180f8e77/tools/k4aviewer/k4astaticimageproperties.h#L88-L102

And with this function to limit the range: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/2feb3425259bf803749065bb6d628c6c180f8e77/tools/k4aviewer/k4adepthpixelcolorizer.h#L87

DOG-j, 2023-02-13:
Hi, thank you for replying this issue.

And I wanna ask exampleRecording.py code is working?

ibaiGorordo, 2023-02-17:
The recording is working well for me. There was an issue with the playback, but it should be fine now. If not, please create a new issue with some more info about what issues you are having.

closed 2023-02-17 by ibaiGorordo

DOG-j, 2023-02-17:
Awesome, thanks!