ibaiGorordo/pyKinectAzure #94: Delay when capturing color images

jonas-hurst, 2023-02-09:
There seems to be a small delay when capturing color camera images of around 1/3 of a second.
Is there a way to get the color images (almost) instantly?

_________
(This problem has driven me crazy for weeks. I finally found a solution, so I am writing this issue to document a solution for everyone to profit. Solution in comment below.)

jonas-hurst, 2023-02-09:
The issue occurs when device color format is set to `device_config.color_format = pykinect.K4A_IMAGE_FORMAT_COLOR_MJPG`. This is the default setting. Decoding the jpg seems to take some time causing the delay.

When using BGRA32 color format instead, no decoding of the image is necessary.
`device_config.color_format = pykinect.K4A_IMAGE_FORMAT_COLOR_BGRA32`

A conversion between from BGRA to BGR color space is however necessary to show the image with opencv's `cv2.imshow()` method.

`
color_image_bgr = cv2.cvtColor(color_image_bgra, cv2.COLOR_BGRA2BGR)
`

closed 2023-02-09 by jonas-hurst

miraclejzd, 2023-04-01:
Hi.
I tried your solution, and foundd it really efficient!

But there is another question: The color of the image are changed, such as the face of the human turns to blue, which is really strange. Do you know how to resolve it?

jonas-hurst, 2023-04-01:
Did you convert to BGR color format? Or do you use an image viewer that does not expect BGR but RGB?

miraclejzd, 2023-04-02:
Yeah, I use ```cv2.cvtColor()``` to convert it to BGR format. And then I use ```cv2.imshow()``` to show the image. Here is part of my code:
```
    device_config = pykinect.default_configuration
    device_config.color_format = pykinect.K4A_IMAGE_FORMAT_COLOR_BGRA32
    device_config.color_resolution = pykinect.K4A_COLOR_RESOLUTION_1080P
    device_config.depth_mode = pykinect.K4A_DEPTH_MODE_WFOV_2X2BINNED

    device = pykinect.start_device(config=device_config)

    cv2.namedWindow('Color image', cv2.WINDOW_NORMAL)
    while True:
        capture = device.update()

        ret, color_image = capture.get_color_image()
        if not ret:
            continue

        color_image = cv2.cvtColor(color_image, cv2.COLOR_RGBA2BGR)
        cv2.imshow('Color image with skeleton', color_skeleton)
```