ibaiGorordo/pyKinectAzure #95: Joint use of multiple kinects

KungZell, 2023-02-18:
Hello, @alfredplpl 
Thank you very much for what a great open source project, I recently noticed that you will be generating 3D skeleton visualization, I would like to ask when it will go live and also if it will be multi-camera linkage? Or will you be producing joint detection with multiple Kinect cameras in multiple views in the future?

ibaiGorordo, 2023-02-19:
Hi, regarding the 3D skeleton, it should be easy since the Open3D support is already there. I will try to get it done next Friday.

About the multi-camera, it is a bit harder. I haven't ported much of the necessary code for it, mainly because I only have one device. Also, it might require quite some work to get what you need, but I probably can't do it with the time I put into it.

KungZell, 2023-02-21:
Okay. Thank you very much for your help.

closed 2023-02-21 by KungZell