ibaiGorordo/pyKinectAzure #100: AttributeError: 'Tracker' object has no attribute '_handle'

bohanfeng, 2023-04-07:
Hello, I encountered this issue while running your code. I hope to receive your answer
My system is ubuntu 20.04 without GPU

user@user-TGL:~/Desktop/pyKinectAzure/examples$ python exampleBodyTrackingColorCamera.py
/workspace/onnx/onnxruntime/onnxruntime/core/providers/cuda/cuda_call.cc:122 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] /workspace/onnx/onnxruntime/onnxruntime/core/providers/cuda/cuda_call.cc:116 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] CUDA failure 35: CUDA driver version is insufficient for CUDA runtime version ; GPU=32766 ; hostname=user-TGL ; expr=cudaSetDevice(info_.device_id); 

[2023-04-08 03:50:10.227] [error] [t=140172] [K4ABT] ../src/TrackerHost/TrackerHost.cpp (184): Create(). /workspace/onnx/onnxruntime/onnxruntime/core/providers/cuda/cuda_call.cc:122 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] /workspace/onnx/onnxruntime/onnxruntime/core/providers/cuda/cuda_call.cc:116 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] CUDA failure 35: CUDA driver version is insufficient for CUDA runtime version ; GPU=32766 ; hostname=user-TGL ; expr=cudaSetDevice(info_.device_id); 

[2023-04-08 03:50:10.227] [error] [t=140172] [K4ABT] ../src/sdk/k4abt.cpp (39): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
Body tracker initialization failed!
  File "exampleBodyTrackingColorCamera.py", line 20, in <module>
    bodyTracker = pykinect.start_body_tracker()
  File "/home/user/.local/lib/python3.8/site-packages/pykinect_azure/pykinect.py", line 57, in start_body_tracker
    return Tracker(Device.calibration, model_type)
  File "/home/user/.local/lib/python3.8/site-packages/pykinect_azure/k4abt/tracker.py", line 12, in __init__
    self._handle = self.create(model_type)
  File "/home/user/.local/lib/python3.8/site-packages/pykinect_azure/k4abt/tracker.py", line 68, in create
    _k4abt.VERIFY(_k4abt.k4abt_tracker_create(self.calibration.handle(), tracker_config, tracker_handle), "Body tracker initialization failed!")
  File "/home/user/.local/lib/python3.8/site-packages/pykinect_azure/k4abt/_k4abt.py", line 183, in VERIFY
    traceback.print_stack()
Exception ignored in: <function Tracker.__del__ at 0x7f03d2848310>
Traceback (most recent call last):
  File "/home/user/.local/lib/python3.8/site-packages/pykinect_azure/k4abt/tracker.py", line 17, in __del__
    self.destroy()
  File "/home/user/.local/lib/python3.8/site-packages/pykinect_azure/k4abt/tracker.py", line 29, in destroy
    if self.is_valid():
  File "/home/user/.local/lib/python3.8/site-packages/pykinect_azure/k4abt/tracker.py", line 20, in is_valid
    return self._handle
AttributeError: 'Tracker' object has no attribute '_handle'

ibaiGorordo, 2023-04-08:
Hi, 

I don't have an Ubuntu system without GPU, so I cannot reproduce the error. I have fixed the default configuration so that if no CUDA is found it will run in CPU mode. But I am not sure if it will work. Let me know if it fixes the issue.

Make sure to upgrade the pykinect-azure python package `pip3 install pykinect-azure --upgrade`

bohanfeng, 2023-04-08:
Wow, amazing! It works!
Thank you so much for your response. Your help is incredibly valuable to me. Wishing you all the best!

closed 2023-04-08 by ibaiGorordo