ibaiGorordo/pyKinectAzure #102: How can I save colored point clouds as .pcd or .ply by depthimage and colorimage?

LiyuRen00, 2023-04-20:


panhigh, 2023-04-26:
same question

panhigh, 2023-04-26:
i try to save by open3d, but the cloud is inaccurate,  so try pykinect, 
can you share your code to my email :panhigh@live.com

Ahyuan233, 2023-04-28:
same question, get_transformed_color_image() returns empty image with all zeros

675506026, 2024-03-21:
This seems to work，But I don't know if he's right
`import pykinect_azure as pykinect
import open3d as o3d
import cv2
import time
if __name__ == "__main__":

    # Initialize the library, if the library is not found, add the library path as argument
    pykinect.initialize_libraries()

    # Modify camera configuration
    device_config = pykinect.default_configuration
    device_config.color_format = pykinect.K4A_IMAGE_FORMAT_COLOR_BGRA32
    device_config.color_resolution = pykinect.K4A_COLOR_RESOLUTION_720P
    device_config.depth_mode = pykinect.K4A_DEPTH_MODE_NFOV_2X2BINNED
    device = pykinect.start_device(config=device_config)
    pcd = o3d.geometry.PointCloud()
    while True:
        starttime = time.time()
        # Get capture
        capture = device.update()

        # Get the 3D point cloud
        ret_point, points = capture.get_transformed_pointcloud()

        # Get the color image in the depth camera axis
        ret_color, color_image = capture.get_color_image()

        if ret_point:
            print("点采集完成")
        if ret_color:
            print('颜色采集完成')
        endtime = time.time()
        if ret_color and ret_point:
            pcd.points =  o3d.utility.Vector3dVector(points)
            colors = cv2.cvtColor(color_image, cv2.COLOR_BGRA2RGB).reshape(-1, 3) / 255
            pcd.colors = o3d.utility.Vector3dVector(colors)
            o3d.io.write_point_cloud('point_cloud1.pcd', pcd)
            break
    device.stop_cameras()

    print('采集时间为',round(endtime-starttime,2),'s')`