ibaiGorordo/pyKinectAzure #107: how to change color image and transformed color image to bgra32 format?

ERICXUCHI, 2023-05-18:
hi, when I run examplePlayback.py, this line "ret_color, color_image = capture.get_transformed_color_image()" would generate error: `[error] [t=20928] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (1253): k4a_transformation_color_image_to_depth_camera(). Require color image and transformed color image both have bgra32 format.`
Can anyone tell me how to convert color image and transformed color image to bgra32 format?
Thank you very much!

lienzzzz, 2023-07-15:
pykinect_azure\k4a\transformation.py  line 74. Transformation.color_image_to_depth_camera()
add
`color_image = Image.create(_k4a.K4A_IMAGE_FORMAT_COLOR_BGRA32, color_image.width, color_image.height, color_image.stride)`
before the code
`_k4a.k4a_transformation_color_image_to_depth_camera(self._handle, depth_image.handle(), color_image.handle(), transformed_color_image.handle())`

lienzzzz, 2023-07-17:
in pykinect_azure/k4a/transformation.py
`
	def color_image_to_depth_camera(self, depth_image, color_image):
		if not depth_image.is_valid() or not color_image.is_valid():
			return Image()

		transformed_color_image = Image.create(_k4a.K4A_IMAGE_FORMAT_COLOR_BGRA32,
												self.depth_resolution.width,
												self.depth_resolution.height,
												self.depth_resolution.width*4)

		# transform color_image's format 2 BGRA32
		# import cv2
		bgra_image_handle = _k4a.k4a_image_t()
		_, m = color_image.to_numpy()
		bgra = cv2.cvtColor(m, cv2.COLOR_BGR2BGRA)
		image_format = _k4a.K4A_IMAGE_FORMAT_COLOR_BGRA32
		buffer = bgra.ctypes.data_as(ctypes.POINTER(ctypes.c_uint8))
		_k4a.VERIFY(_k4a.k4a_image_create_from_buffer(image_format, bgra.shape[1], bgra.shape[0], bgra.shape[1]*4, buffer, bgra.nbytes, ctypes.c_void_p(0), ctypes.c_void_p(0), bgra_image_handle), "JPG to BGRA32 ERROR")

		_k4a.k4a_transformation_color_image_to_depth_camera(self._handle, depth_image.handle(), bgra_image_handle, transformed_color_image.handle())

		return transformed_color_image
`