ibaiGorordo/pyKinectAzure #115: Set up the project again

tyAlexios, 2024-01-12:
if I change some codes in device.py and capture.py, do I have to set up this project again to make the changes take effect? And how to set it up again? Thanks

zhonghong322, 2024-01-15:
1、python setup.py build 
2、python setup.py install

zhonghong322, 2024-01-15:
Due to the end-of-life (EOL) status of Azure Kinect, which is no longer maintained, Orbbec company provides Femto Mega and Femto Bolt as replacements that are secure and compatible with the Azure Kinect interface. I have noticed that many former users have been using the pyKinectAzure repository. We have forked this repository and made some simple adaptations to support Femto Mega and Femto Bolt.